hbldh/bleak #1590: WinError The operation was canceled by the user

wencyxie, 2024-06-03:
* bleak version: 0.22.1
* Python version: 3.9.12
* Operating System: Windows 11 Enterprise
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I tried to read the characteristic, and whenever it reads, it says "services changed" "unhandled services changed event". Finally there is an error "OSError: [WinError -2147023673] The operation was canceled by the user"

### What I Did
```
async def test():
    logger.info("start scanning...")
    device = await bs.find_device_by_name('xxx')
    logger.info(device)
    u.register_uuids(uuid_dict)
    async with bc(device) as client:
        char = client.services.get_characteristic('xxx')
        logger.info(char)
        value = await client.read_gatt_char(char)
        print(value)
```

### Logs
```
2024-06-03 17:37:34,677 bleak.backends.winrt.scanner DEBUG 89 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
2024-06-03 17:37:34,678 logger INFO C6:AA:CC:93:56:AA: ProSmart 96F0
2024-06-03 17:37:34,682 bleak.backends.winrt.client DEBUG Connecting to BLE device @ C6:AA:CC:93:56:AA     
2024-06-03 17:37:34,835 bleak.backends.winrt.client DEBUG getting services (service_cache_mode=None, cache_mode=None)...
2024-06-03 17:37:35,896 bleak.backends.winrt.client DEBUG session_status_changed_event_handler: id: BluetoothLE#BluetoothLE74:04:f1:47:bc:b4-c6:aa:cc:93:56:aa, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>
2024-06-03 17:37:35,925 bleak.backends.winrt.client DEBUG max_pdu_size_changed_handler: 247
2024-06-03 17:37:36,037 logger INFO 6cd6c8b5-e378-0106-010a-1b9740683449 (Handle: 15): Device Information
2024-06-03 17:37:36,859 bleak.backends.winrt.client DEBUG C6:AA:CC:93:56:AA: services changed
2024-06-03 17:37:36,859 bleak.backends.winrt.client WARNING C6:AA:CC:93:56:AA: unhandled services changed event
2024-06-03 17:37:36,946 bleak.backends.winrt.client DEBUG Disconnecting from BLE device...
2024-06-03 17:37:42,291 bleak.backends.winrt.client DEBUG max_pdu_size_changed_handler: 23
2024-06-03 17:37:42,292 bleak.backends.winrt.client DEBUG session_status_changed_event_handler: id: BluetoothLE#BluetoothLE74:04:f1:47:bc:b4-c6:aa:cc:93:56:aa, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.CLOSED: 0>
2024-06-03 17:37:42,292 bleak.backends.winrt.client DEBUG closing requester
2024-06-03 17:37:42,292 bleak.backends.winrt.client DEBUG closing session
Traceback (most recent call last):
  File "c:\Users\85194112\OneDrive - BAT\Automation\Robot_framework\Demo\bleak_demo.py", line 133, in <module>
    asyncio.run(test())
  File "c:\Users\85194112\AppData\Local\Programs\Python\Python39\lib\asyncio\runners.py", line 44, in run  
    return loop.run_until_complete(main)
  File "c:\Users\85194112\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "c:\Users\85194112\OneDrive - BAT\Automation\Robot_framework\Demo\bleak_demo.py", line 126, in test 
    value = await client.read_gatt_char(char)
  File "c:\Users\85194112\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\__init__.py", line 721, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "c:\Users\85194112\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 859, in read_gatt_char
    await characteristic.obj.read_value_async(
OSError: [WinError -2147023673] The operation was canceled by the user
```

I don't think that I did anything to cancel the operation. And I also try write and notify, the failure is the same as read.

dlech, 2024-06-03:
Could you also capture Bluetooth packets ad described in the troubleshooting page of the docs and share that?

wencyxie, 2024-06-04:
I tried to scan and log the advertisement data while running the test script, but I am not sure if the new log shared below is what you asked from me. Please help to confirm, thanks a lot!
```
2024-06-04 10:36:25,988 logger INFO start scanning...
2024-06-04 10:36:26,035 logger INFO C6:AA:CC:93:56:AA: ProSmart 96F0:AdvertisementData(local_name='ProSmart 96F0', manufacturer_data={38182: b'\xaaV\x93\xcc\xaa\xc6'}, rssi=-81)
2024-06-04 10:36:26,035 logger INFO C6:AA:CC:93:56:AA: ProSmart 96F0:AdvertisementData(local_name='ProSmart 96F0', manufacturer_data={38182: b'\xaaV\x93\xcc\xaa\xc6'}, service_uuids=['6cd6c8b5-e378-0106-000a-1b9740683449'], rssi=-81)
2024-06-04 10:36:27,801 logger INFO C6:AA:CC:93:56:AA: ProSmart 96F0:AdvertisementData(local_name='ProSmart 96F0', manufacturer_data={38182: b'\xaaV\x93\xcc\xaa\xc6'}, service_uuids=['6cd6c8b5-e378-0106-000a-1b9740683449'], rssi=-78)
2024-06-04 10:36:27,801 logger INFO C6:AA:CC:93:56:AA: ProSmart 96F0
2024-06-04 10:36:28,717 logger INFO C6:AA:CC:93:56:AA: ProSmart 96F0:AdvertisementData(local_name='ProSmart 96F0', manufacturer_data={38182: b'\xaaV\x93\xcc\xaa\xc6'}, service_uuids=['6cd6c8b5-e378-0106-000a-1b9740683449'], rssi=-83)
2024-06-04 10:36:29,324 logger INFO 6cd6c8b5-e378-0106-010a-1b9740683449 (Handle: 15): Device Information
2024-06-04 10:36:29,636 logger INFO C6:AA:CC:93:56:AA: ProSmart 96F0:AdvertisementData(local_name='ProSmart 96F0', manufacturer_data={38182: b'\xaaV\x93\xcc\xaa\xc6'}, service_uuids=['6cd6c8b5-e378-0106-000a-1b9740683449'], rssi=-97)
Traceback (most recent call last):
  File "c:\Users\85194112\OneDrive - BAT\Automation\Robot_framework\Demo\bleak_demo.py", line 152, in <module>
    loop.run_until_complete(main())
  File "c:\Users\85194112\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "c:\Users\85194112\OneDrive - BAT\Automation\Robot_framework\Demo\bleak_demo.py", line 144, in main 
    await asyncio.gather(scan(), test())
  File "c:\Users\85194112\OneDrive - BAT\Automation\Robot_framework\Demo\bleak_demo.py", line 129, in test 
    value = await client.read_gatt_char(char)
  File "c:\Users\85194112\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\__init__.py", line 721, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "c:\Users\85194112\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 859, in read_gatt_char
    await characteristic.obj.read_value_async(
OSError: [WinError -2147023673] The operation was canceled by the user
```

dlech, 2024-06-04:
Nope, I meant that you should capture Bluetooth packets with Wireshark. This is the same log as before.