hbldh/bleak #1595: unable to use start_notify

gilcu3, 2024-06-09:
* bleak version: 0.22.1
* Python version: 3.12.3
* Operating System: Archlinux
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.76

### Description

I am trying to connect to a smart scale. There is a single characteristic that has notifications enabled, and I am using as a starter the `enable_notifications.py` example from the repo.

### What I Did

```
python enable_notifications.py --address dc:23:4d:2c:4e:b6 -d 00002b10-0000-1000-8000-00805f9b34fb
```
Using `wireshark` I see that the last packet sent is always:
```
Bluetooth Attribute Protocol
    Opcode: Read Request (0x0a)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..00 1010 = Method: Read Request (0x0a)
    Handle: 0x0009 (Generic Access: Peripheral Preferred Connection Parameters)
        [Service UUID: Generic Access (0x1800)]
        [UUID: Peripheral Preferred Connection Parameters (0x2a04)]
```
for which the device never replies nor sends anything else.
### Logs

[log.txt](https://github.com/user-attachments/files/15751470/log.txt)

dlech, 2024-06-09:
This error is coming from the peripheral. I would recommend sniffing Bluetooth packets to see how the official app communicates and compare that to what Bleak is doing.

gilcu3, 2024-06-09:
I did already, the most notable difference I can see is that a few generic packets are not sent, therefore my suspicious that one of those is not supported and blocks communication.
[bt.pcap](https://github.com/user-attachments/files/15753104/bt.pcap.txt)

dlech, 2024-06-09:
I take it that `bt.pcap` is the working case from a non-Linux system?

If the peripheral says it has the Peripheral Preferred Connection Parameters characteristic but fails to respond to read requests, I'm not sure what could be done.

gilcu3, 2024-06-10:
Yes, it is the working case from Android. I think the device might have wrong characteristics, is it possible in `bleak` to disable some of that default behavior?