hbldh/bleak #1617: cannot connect again unless reboot

hopethis, 2024-07-16:
* bleak version: 0.22.2
* Python version: 3.11.3
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Bleak works fine the first time after Booting windows. Buf after one successfull operation I cant get the same programm to run. Either the connection cannot be established or it wasnt properly closed the first time. 

import logging
from bleak import BleakClient

logging.basicConfig(level=logging.DEBUG)

address = "0D:8C:DC:22:11:11"
async def connect():
    async with BleakClient(address, timeout=30.0) as client:
        services = await client.get_services()
        print(services)

import asyncio
asyncio.run(connect())

After first execution the status toggles between active and closed
Error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>
error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.CLOSED: 0>

and failes at the end 

ile "C:\Users\ladmin\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 168, 
in _ensure_success
    raise BleakError(f"{fail_msg}: Unreachable")
bleak.exc.BleakError: Could not get GATT services: Unreachable

A reboot fixes the problem and gives another try. .... but this is really annoying...

Does anybody has an idea?

dlech, 2024-07-16:
Sounds like a bad Windows Bluetooth driver. You could try logging Bluetooth packets to see what is going on. See troubleshooting page in the docs for instructions.

hopethis, 2024-07-16:
Hi,

yes it can be the driver. I checked for updates but there is non available...

i did already but couldnt upload pcap files here. 

1	0.000000	host	controller	HCI_CMD	10	Sent LE Set Extended Scan Enable
2	0.002114	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
3	0.002345	host	controller	HCI_CMD	6	Sent Vendor Command 0x001E (opcode 0xFC1E)
4	0.003991	controller	host	HCI_EVT	8	Rcvd Command Complete (Vendor Command 0x001E [opcode 0xFC1E])
5	0.004106	host	controller	HCI_CMD	12	Sent LE Set Extended Scan Parameters
6	0.006003	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Parameters)
7	0.006112	host	controller	HCI_CMD	10	Sent LE Set Extended Scan Enable
8	0.007996	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
9	0.008123	host	controller	HCI_CMD	16	Sent Vendor Command 0x002B (opcode 0xFC2B)
10	0.011000	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
11	0.011133	host	controller	HCI_CMD	16	Sent Vendor Command 0x002B (opcode 0xFC2B)
12	0.013001	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
13	0.013052	host	controller	HCI_CMD	16	Sent Vendor Command 0x002B (opcode 0xFC2B)
14	0.015033	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
15	0.015096	host	controller	HCI_CMD	18	Sent Vendor Command 0x002B (opcode 0xFC2B)
16	0.017011	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Parameters)
17	0.017249	host	controller	HCI_CMD	18	Sent Vendor Command 0x002B (opcode 0xFC2B)
18	0.019014	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Parameters)
19	0.019166	host	controller	HCI_CMD	18	Sent Vendor Command 0x002B (opcode 0xFC2B)
20	0.021016	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Parameters)
21	0.021175	host	controller	HCI_CMD	16	Sent Vendor Command 0x002B (opcode 0xFC2B)
22	0.023030	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
23	0.023179	host	controller	HCI_CMD	16	Sent Vendor Command 0x002B (opcode 0xFC2B)
24	0.026020	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
25	0.026172	host	controller	HCI_CMD	16	Sent Vendor Command 0x002B (opcode 0xFC2B)
26	0.029025	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
27	0.030019	controller	host	HCI_EVT	47	Rcvd LE Meta (LE Extended Advertising Report)
28	0.031037	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
29	0.033003	controller	host	HCI_EVT	47	Rcvd LE Meta (LE Extended Advertising Report)
30	0.034015	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
31	0.147086	controller	host	HCI_EVT	46	Rcvd LE Meta (LE Extended Advertising Report)
32	0.160019	controller	host	HCI_EVT	48	Rcvd LE Meta (LE Extended Advertising Report)
33	0.161018	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
34	0.396133	controller	host	HCI_EVT	37	Rcvd LE Meta (LE Extended Advertising Report)
35	0.500047	controller	host	HCI_EVT	59	Rcvd LE Meta (LE Extended Advertising Report)
36	0.503016	controller	host	HCI_EVT	56	Rcvd LE Meta (LE Extended Advertising Report)
37	0.504009	controller	host	HCI_EVT	52	Rcvd LE Meta (LE Extended Advertising Report)
38	0.505005	controller	host	HCI_EVT	47	Rcvd LE Meta (LE Extended Advertising Report)
39	0.505998	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
40	0.522022	controller	host	HCI_EVT	46	Rcvd LE Meta (LE Extended Advertising Report)
41	0.738947	controller	host	HCI_EVT	46	Rcvd LE Meta (LE Extended Advertising Report)
42	0.739959	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
43	0.751017	controller	host	HCI_EVT	56	Rcvd LE Meta (LE Extended Advertising Report)
44	0.752002	controller	host	HCI_EVT	52	Rcvd LE Meta (LE Extended Advertising Report)
45	0.852097	controller	host	HCI_EVT	57	Rcvd LE Meta (LE Extended Advertising Report)
46	0.860015	controller	host	HCI_EVT	47	Rcvd LE Meta (LE Extended Advertising Report)
47	0.860993	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
48	0.864995	controller	host	HCI_EVT	47	Rcvd LE Meta (LE Extended Advertising Report)
49	0.865992	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
50	0.974128	controller	host	HCI_EVT	53	Rcvd LE Meta (LE Extended Advertising Report)
51	0.984015	controller	host	HCI_EVT	48	Rcvd LE Meta (LE Extended Advertising Report)
52	0.984994	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
53	1.223313	controller	host	HCI_EVT	48	Rcvd LE Meta (LE Extended Advertising Report)
54	1.224019	controller	host	HCI_EVT	50	Rcvd LE Meta (LE Extended Advertising Report)
55	1.331092	controller	host	HCI_EVT	47	Rcvd LE Meta (LE Extended Advertising Report)
56	1.332007	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
57	1.333030	controller	host	HCI_EVT	46	Rcvd LE Meta (LE Extended Advertising Report)
58	1.336000	controller	host	HCI_EVT	56	Rcvd LE Meta (LE Extended Advertising Report)
59	1.336997	controller	host	HCI_EVT	52	Rcvd LE Meta (LE Extended Advertising Report)
60	1.343010	controller	host	HCI_EVT	46	Rcvd LE Meta (LE Extended Advertising Report)
61	1.483119	controller	host	HCI_EVT	44	Rcvd LE Meta (LE Extended Advertising Report)
62	1.484014	controller	host	HCI_EVT	29	Rcvd LE Meta (LE Extended Advertising Report)
[... cut ...]
290	21.876759	host	controller	HCI_CMD	18	Sent Vendor Command 0x002B (opcode 0xFC2B)
291	21.878454	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Parameters)
292	21.878497	host	controller	HCI_CMD	16	Sent Vendor Command 0x002B (opcode 0xFC2B)
293	21.880433	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
294	21.880580	host	controller	HCI_CMD	16	Sent Vendor Command 0x002B (opcode 0xFC2B)
295	21.894473	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)

dlech, 2024-07-16:
FYI, you can put them in a `.zip` file and upload them. But I think we can see enough info from the pasted log.

```console
269	17.834600	localhost ()	Suunto_3e:57:61 (Movesense 214230001254)	ATT	14	Sent Write Request, Handle: 0x000d (Unknown: Service Changed: Client Characteristic Configuration)
```

There is no reply from this request, so it looks like you have a defective Bluetooth peripheral.

You might be able to avoid this by trying the `use_cached_services=False` argument. See https://bleak.readthedocs.io/en/latest/backends/windows.html#bleak.backends.winrt.client.WinRTClientArgs.use_cached_services.

hopethis, 2024-07-17:
Thanks, 
whats odd is that it seems to work with other computer pretty well and everytime I reboot the machine its working as well. Passing the option used cached services to False unfortunally didnt solve the issue.

dlech, 2024-07-17:
You could do a packet capture of the working cases and compare them to the not working cases to see what the difference is.

Aeva, 2024-08-03:
This started happening for me recently on Linux. 

Bleak: 0.22.2
Python: 3.12.4
OS: Fedora 40
bluetoothctl: 5.77

The version of bleak I have installed was through pip.  This may have been due to me switching from NixOS to Fedora.  The version of bleak I had on NixOS was installed via some nix magic, whereas the version of bleak I have installed now was just the one available on pip.  I don't recall if this ever worked correctly on Fedora / with pip version.

This is the error that prints the second time I try to run my program after a successful first run:

```
Traceback (most recent call last):
  File "/home/aeva/.local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 273, in connect
    await self.get_services(
  File "/home/aeva/.local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 661, in get_services
    self.services = await manager.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aeva/.local/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 666, in get_services
    await self._wait_for_services_discovery(device_path)
  File "/home/aeva/.local/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 783, in _wait_for_services_discovery
    done, _ = await asyncio.wait(
              ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/asyncio/tasks.py", line 464, in wait
    return await _wait(fs, timeout, return_when, loop)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/asyncio/tasks.py", line 550, in _wait
    await waiter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib64/python3.12/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib64/python3.12/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/aeva/science/autonomia/autonomia/bluetooth.py", line 110, in heart_monitor_proc
    asyncio.run(heart_monitor_runner(command_queue, event_queue, device_address))
  File "/usr/lib64/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/aeva/science/autonomia/autonomia/bluetooth.py", line 69, in heart_monitor_runner
    async with BleakClient(device_address) as client:
  File "/home/aeva/.local/lib/python3.12/site-packages/bleak/__init__.py", line 570, in __aenter__
    await self.connect()
  File "/home/aeva/.local/lib/python3.12/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aeva/.local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 151, in connect
    async with async_timeout(timeout):
  File "/usr/lib64/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

dlech, 2024-08-09:
Oof. This is happening to me too. I haven't had time to dig into it much yet though.

Ubuntu 24.04, BlueZ 5.72.

The only thing that I know of that has changed for me is updating to Ubuntu 24.04, so I'm suspecting a bug in BlueZ at this point.

dlech, 2024-08-09:
Forgot to mention... better workaround is to just restart BlueZ. 

    sudo systemctl restart bluetooth.service

dlech, 2024-08-10:
> The only thing that I know of that has changed for me is updating to Ubuntu 24.04, so I'm suspecting a bug in BlueZ at this point.

Tried with Ubuntu 20.04 and BlueZ 5.64 in a VM and it is still broken, so likely not a regression in BlueZ.

dlech, 2024-08-11:
> This started happening for me recently on Linux.

This issue is about Windows, not Linux. So please start a new issue for the Linux issue and include a full reproducible test case.