hbldh/bleak #1630: 'NSKVONotifying_CBCentralManager' object has no attribute 'retrieveAddressForPeripheral_'

CoderXL1, 2024-08-18:
* bleak version: 0.22.2
* Python version: 3.12.5
* Operating System: Mac OS 11 Big Sur
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

When running home assistant, an error occurs: 
AttributeError: 'NSKVONotifying_CBCentralManager' object has no attribute 'retrieveAddressForPeripheral_'. Did you mean: 'retrieveConnectedPeripherals'?

CoderXL1, 2024-08-18:
python3.12/site-packages/bleak/backends/corebluetooth/scanner.py, line 134, in callback self._manager.central_manager.retrieveAddressForPeripheral_(p)

dlech, 2024-08-18:
Duplicate of https://github.com/hbldh/bleak/issues/1286. But I guess we can leave this one open since that one is closed. Same comments still apply though.