hbldh/bleak #1634: Bleak pairing not working gives Failed to pair: org. bluez.Error.AuthenticationRejected error on raspberry pi

wahadatjan, 2024-08-27:
bleak version: 0.22.2
Python version:3.11.6
BlueZ version:5.68

### Description

I used bleak library to pair the ble devices but get error like

  | Attempting to pair with EA:A0:01:2B:32:A9
-- | --
  | Failed to pair: org. bluez.Error.AuthenticationRejected

`

### What I Did

`.          import asyncio
               
from bleak import BleakScanner, BleakClient

     Wahdat_name = "SM-DOM2"
     characteristic_uuid = "00002A37-0000-1000-8000-00805f9b34fb"
     async def scan_and_pair():
    # Scan for devices
    print("Scanning for BLE devices...")
    devices = await BleakScanner.discover()
    
    if not devices:
        print("No BLE devices found.")
        return
    
    # List found devices
    print("Found devices:")
    for i, device in enumerate(devices):
        print(f"{i}: {device.name} - {device.address}")

    # Check if Wahdat's Android Watch is found by name
    device_found = False
    for device in devices:
        if device.name == Wahdat_name:
            device_found = True
            selected_device = device.address
            break
    
    print("SELECTED",selected_device)
    if not device_found:
        print(f"{Wahdat_name} not found.")
        return
    
    # Pair with Wahdat's Android Watch
    async with BleakClient(selected_device) as client:
        try:
            print(f"Attempting to pair with {selected_device} ({Wahdat_name})...")
            paired = await client.pair()
            if paired:
                value = await client.read_gatt_char(characteristic_uuid)
                print("Read value:", value)
                print(f"Successfully paired with {selected_device} ({Wahdat_name})")
            else:
                print(f"Failed to pair with {selected_device} ({Wahdat_name})")
        except Exception as e:
            print(f"Error during pairing: {e}")

asyncio.run(scan_and_pair())

`

dlech, 2024-08-27:
If you run `agent on` in `bluetootctl` before `pair ...` does it prompt you for a passkey?

wahadatjan, 2024-08-28:
@dlech agent on and while pairing still getting same error 
[image]

dlech, 2024-08-28:
If BlueZ can't pair, then Bleak isn't going to be able to either. You could try logging Bluetooth packets to see what is going on behind the scenes to try to figure out what the actual problem is.