hbldh/bleak #1643: Support Android with Flet

gunaNeelamegam, 2024-09-28:
# Cannot able to you bleak with flet.

* While import based platform in bleak/__init__.py condition env is not exists.

* While build with flet the env is not exist. Need's to remove make as generic.

dlech, 2024-10-05:
I didn't see anything in Flet that allows calling android APIs, so this doesn't seem like it would be feasible at this point in time.

gunaNeelamegam, 2024-10-06:
But, they referred we can able to  compile and include the .so file. I
tried with different approach I think we need to dynamically load the
custom java class right now in the blesk it's static for using the custom
java classes.

https://flet.dev/docs/publish/android/

dlech, 2024-10-06:
Ok, so Flet is using Python4Android. So it seems like it should just work then. 

>  condition env is not exists.

Are you saying that the `P4A_BOOTSTRAP` environment variable is not set?

gunaNeelamegam, 2024-10-06:
Not only that we need use the java custom classes based on project package
right now it's static.