hbldh/bleak #1645: macOS client.start_notify fails after reconnect

dakhnod, 2024-09-28:
* bleak version: 0.22.2
* Python version: 3.9
* Operating System: macOS sonoma 14.4.1
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

When running `client.start_notify` on a device that has previously been connected to and that disconnected and connected again, it throws the exception `ValueError: Characteristic notifications already started`.

Not calling the function a second time causes no notifications to arrive.

dlech, 2024-09-29:
Can you give an example program and the exact steps to reproduce the problem?

dakhnod, 2024-10-02:
Yep, sorry. Attached is a minimal demo program.
Somehow, bleak or some other component seems to remember a notification subscription, even if the device disconnected...

```py
import bleak
import asyncio

# replace this with anything suiting on your device
notify_uuid = '00002a56-0000-1000-8000-00805f9b34fb'

async def main():
    scanner = bleak.BleakScanner()

    disconnect_event = asyncio.Event()
    client = bleak.BleakClient(
        await scanner.find_device_by_name('BLEnky demo'),
        lambda client: disconnect_event.set()
    )
    await client.connect()
    await client.start_notify(notify_uuid, print)

    print('connected, please restart device')

    await disconnect_event.wait()

    print('reconnecting...')

    await client.connect()
    # this will throw on macOS, not on linux
    await client.start_notify(notify_uuid, print)

    print('reconnected.')

asyncio.run(main())
```

Steps to reproduce:
1. Put in the name of a test device you can connect to and restart
2. Put in the UUID of some notification that can be subscribed to
3. Run script
4. When prompted, restart device
5. Watch exception get thrown, but only on mac

Besides, great work on this! This library is a pleasure to use, I wish there was something comperable for js! Thank you!

MAKOMO, 2024-10-06:
I can duplicate this also for v0.22.3 on macOS 15.0.1

dlech, 2024-10-06:
Calling `connect()` a second time on the same `BleakClient` object is something that has never worked well in Bleak. We recommend to create a new `BleakClient` object in order to connect again.

MAKOMO, 2024-10-07:
Thanks David for this hint! Not reusing that client resolves the reconnect issues for me. Keep going!

dlech, 2025-05-10:
Closing since it sounds like we have a solution.

closed 2025-05-10 by dlech