hbldh/bleak #1647: DBus.Error.AccessDenied

ThomasJF, 2024-09-30:
* bleak version: 0.22.2
* Python 3.9.2:
* Raspian Bullseye

Error Message:
python bluetooth org.freedesktop.DBus.Error.AccessDenied] Client tried to send a message other than Hello without being registered

**Here is what I do:**

```
adr="34:B7:DA:65:4C:F9"                                                                                   # Device Address
device_control_ZERO = "adbc045e-09e8-4332-ae0f-1f9dd87b4f4d"               # UUID for ZERO indicator (true or false). 
device_control_COMMAND= "908d8f21-4fda-4954-a5aa-fb94c521a5ee"       # UUID for Command and response

async def Read_Ping():
    global adr,device_control_COMMAND,device_control_ZERO
    async with BleakClient(adr) as client:
        print("Ping. Connected: ",client.is_connected)
        await client.write_gatt_char(device_control_COMMAND, b"Ping", response=False)
        data = await client.read_gatt_char(device_control_COMMAND)        
        print("Response: ",data)
        data = await client.read_gatt_char(device_control_ZERO)
        print("ZERO: ",data)

def MyThread():
    global StopRequest,Start,Online
    print("\n Thread Started!")
    while True:
        try:
            while Start!="TRUE":
                if StopRequest!=0: raise ExitRequest
                try:           
                    print("Ping!")
                    asyncio.run(Read_Ping()) # Check Device is online by run Ping. If the function returns, Device is OK 
                    Online=True
                    print("Online=",Online,"  Start=",Start)
                    pass  
                except Exception as e:
                    Online=False
                    print("Device is Offline ",e)
                    pass
                if StopRequest!=0: break
                time.sleep(3)
               
        #Start working cycle
        Start="FALSE"
        print("Device is available")
```

The thread is an endless Loop. When "Start" is set, it does on cycle of work and then hangs around agin pinging the device.
Now this works for a long time (hours), but at a certain time the program starts to display the error message:
"Client tried to send a message other than Hello without being registered".
So far I did not find out when and under what condition the code stops working.
When I kill the program and restart it, the thing works again.
Any clue what goes on? Thank you.
Thomas

dlech, 2024-09-30:
Bleak does not support calling `asyncio.run()` multiple times. Please rewrite your thread to only call that once so that the asyncio event loop is always running for the entire lifetime of your application.

ThomasJF, 2024-09-30:
Tank you very much for your quick answer.