hbldh/bleak #1657: BleakClient can not run on RISC-V linux

mcpanl, 2024-10-07:
* bleak version: 0.22.3
* Python version: 3.11
* Operating System: RISC-V
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.6.8

### Description
[image]

The following code can run normally on the Windows platform, but it cannot run on Linux development boards on the RISC-V platform.

[image]

### What I Did
1. get a linux borard (Sipeed LicheeRV Nano)
[https://www.alibaba.com/product-detail/Sipeed-LicheeRV-Nano-Litchi-Pi-SG2002_1601059587927.html?spm=a2700.galleryofferlist.normal_offer.d_title.59f213a0npUk9B](https://www.alibaba.com/product-detail/Sipeed-LicheeRV-Nano-Litchi-Pi-SG2002_1601059587927.html?spm=a2700.galleryofferlist.normal_offer.d_title.59f213a0npUk9B)

wiki: [https://wiki.sipeed.com/maixpy/en/index.html](https://wiki.sipeed.com/maixpy/en/index.html)

2. pip install bleak

3. create hci with /dev/ttyS1 and up it.
[image]
[image]

4. run above-mentioned test code

``` python
import asyncio
from bleak import BleakScanner, BleakClient

async def main():
    device = await BleakScanner.find_device_by_address(
            'F6:0C:C0:F5:66:52'
        )

    print(device)

    if device is None:
        print('not found device')
        return

    async with BleakClient(device) as client:
        print('connected')
        print(client)

asyncio.run(main())
```

### Logs
```
F6:0C:C0:F5:66:52: NRF_BATTERY_01
Traceback (most recent call last):
  File "src/dbus_fast/_private/unmarshaller.py", line 301, in dbus_fast._private.unmarshaller.Unmarshaller._read_sock_with_fds
BlockingIOError: [Errno 11] Resource temporarily unavailable

During handling of the above exception, another exception occurred:

BlockingIOError
Exception ignored in: 'dbus_fast._private.unmarshaller.Unmarshaller._read_sock_with_fds'
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
BlockingIOError: 
```

mcpanl, 2024-10-07:
this is full logs
```
# btmon 
Bluetooth monitor ver 5.68
= Note: Linux version 5.10.4-tag- (riscv64)                                                0.708175
= Note: Bluetooth subsystem version 2.22                                                   0.708186
= New Index: C4:3C:B0:6D:53:BD (Primary,UART,hci0)                                  [hci0] 0.708190
= Open Index: C4:3C:B0:6D:53:BD                                                     [hci0] 0.708193
= Index Info: C4:3C:B0:6D:53:BD (RivieraWaves S.A.S)                                [hci0] 0.708198
@ MGMT Open: bluetoothd (privileged) version 1.18                                 {0x0001} 0.708202

@ MGMT Command: Start Service Discovery (0x003a) plen 4                   {0x0001} [hci0] 22.810137
        Address type: 0x06
          LE Public
          LE Random
        RSSI: invalid (0x7f)
        UUIDs: 0
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                       #1 [hci0] 22.811904
        Address: 3D:31:BA:07:FD:92 (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                     #2 [hci0] 22.812883
      LE Set Random Address (0x08|0x0005) ncmd 5
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8             #3 [hci0] 22.812987
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
        PHYs: 0x01
        Entry 0: LE 1M
          Type: Active (0x01)
          Interval: 11.250 msec (0x0012)
          Window: 11.250 msec (0x0012)
> HCI Event: Command Complete (0x0e) plen 4                                     #4 [hci0] 22.813647
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 5
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                 #5 [hci0] 22.813707
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                     #6 [hci0] 22.814640
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 5
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                            {0x0001} [hci0] 22.814764
      Start Service Discovery (0x003a) plen 1
        Status: Success (0x00)
        Address type: 0x06
          LE Public
          LE Random
@ MGMT Event: Discovering (0x0013) plen 2                                 {0x0001} [hci0] 22.814786
        Address type: 0x06
          LE Public
          LE Random
        Discovery: Enabled (0x01)
> HCI Event: LE Meta Event (0x3e) plen 28                                       #7 [hci0] 22.824706
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: 54:B0:00:9A:91:70 (Resolvable)
        Data length: 16
        TX power: -9 dBm
        Service Data: Google (0xfe2c)
          Data: 003000000011b41464
        RSSI: -83 dBm (0xad)
> HCI Event: LE Meta Event (0x3e) plen 28                                       #8 [hci0] 22.825432
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Random (0x01)
        Address: 54:B0:00:9A:91:70 (Resolvable)
        Data length: 16
[... 460 lines cut ...]
        Count: 1
        Latency: 21 msec (21-49 msec ~35 msec)
> ACL Data RX: Handle 7 flags 0x02 dlen 5                                      #64 [hci0] 25.344357
      ATT: Write Response (0x13) len 0
< ACL Data TX: Handle 7 flags 0x00 dlen 7                                      #65 [hci0] 25.345641
      ATT: Read Request (0x0a) len 2
        Handle: 0x0003 Type: Device Name (0x2a00)
> HCI Event: Number of Completed Packets (0x13) plen 5                         #66 [hci0] 25.394379
        Num handles: 1
        Handle: 7 Address: F6:0C:C0:F5:66:52 (OUI F6-0C-C0)
        Count: 1
        Latency: 48 msec (21-49 msec ~42 msec)
> ACL Data RX: Handle 7 flags 0x02 dlen 12                                     #67 [hci0] 25.544456
      ATT: Read Response (0x0b) len 7
< ACL Data TX: Handle 7 flags 0x00 dlen 7                                      #68 [hci0] 25.544907
      ATT: Read Request (0x0a) len 2
        Handle: 0x0005 Type: Appearance (0x2a01)
> HCI Event: Number of Completed Packets (0x13) plen 5                         #69 [hci0] 25.594359
        Num handles: 1
        Handle: 7 Address: F6:0C:C0:F5:66:52 (OUI F6-0C-C0)
        Count: 1
        Latency: 49 msec (21-49 msec ~45 msec)
> ACL Data RX: Handle 7 flags 0x02 dlen 7                                      #70 [hci0] 25.744389
      ATT: Read Response (0x0b) len 2
```

dlech, 2024-10-07:
This looks like an issue with https://github.com/Bluetooth-Devices/dbus-fast, not Bleak.

closed 2025-05-16 by dlech