hbldh/bleak #1672: Method "StartNotify" with signature "" on interface "org.bluez.GattCharacteristic1" doesn't exist

russellproud, 2024-10-23:
* bleak version: 0.22.3 (experienced same on 0.19.5, upgraded to eliminate version)
* Python version: 3.12
* Operating System: Ubuntu 24.04.1
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.72

### Description
I am attempting to connect to a Bluetooth LE device and receive notifications.

I am using the BleakClientBlueZDBus backend client as I am required to negotiate the MTU with this device.

I had previously beeing using BleakClient, though believe (not confirmed) that the MTU was causing issues receiving the notifications.

I am receiving the error (as shown in debug logs below) of ```bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownMethod] Method "StartNotify" with signature "" on interface "org.bluez.GattCharacteristic1" doesn't exist```

### Code to Reproduce
```
async def main(device_mac):
    global client
    
    client = BleakClientBlueZDBus(DEVICE_MAC, adapter="hci1")
    await client.connect()
    print(client.is_connected)
    await client._acquire_mtu()
    
    await handle_notifications()

async def handle_notifications():
    global client    
    while True:
        
            services = await client.get_services()
            target_service = None
            for service in services:
                if target_service:
                    break
                for char in service.characteristics:
                    if char.uuid == SERVICE_UUID:
                        target_service = service
                        break

            await asyncio.sleep(5)
            await client.start_notify(target_service, notification_handler, adapter="hci1")
            print(f"Notification handler started for {NOTIFICATION_UUID}")
```

### Wireshark Capture
[capture.txt](https://github.com/user-attachments/files/17488164/capture.txt)

### Debug Logs
```
2024-10-23 20:09:27,799 bleak.backends.bluezdbus.client MainThread DEBUG: Connecting to device @ F9:82:74:04:59:8B
2024-10-23 20:09:27,803 bleak.backends.bluezdbus.manager MainThread DEBUG: initial properties: {'/org/bluez': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.AgentManager1': {}, 'org.bluez.ProfileManager1': {}, 'org.bluez.HealthManager1': {}}, '/org/bluez/hci1': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Adapter1': {'Address': 'E8:9C:25:59:E3:A9', 'AddressType': 'public', 'Name': 'XPS-15-9500', 'Alias': 'XPS-15-9500', 'Class': 7078156, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0548', 'Roles': ['central', 'peripheral'], 'Manufacturer': 93, 'Version': 10}, 'org.freedesktop.DBus.Properties': {}, 'org.bluez.BatteryProviderManager1': {}, 'org.bluez.GattManager1': {}, 'org.bluez.Media1': {'SupportedUUIDs': ['0000110a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb']}, 'org.bluez.NetworkServer1': {}, 'org.bluez.LEAdvertisingManager1': {'ActiveInstances': 0, 'SupportedInstances': 4, 'SupportedIncludes': ['tx-power', 'appearance', 'local-name'], 'SupportedSecondaryChannels': ['1M', '2M', 'Coded']}}}
2024-10-23 20:09:27,866 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', True)>}, []]
2024-10-23 20:09:27,908 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci1/dev_20_D4_D9_6D_3D_45', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', 20:D4:D9:6D:3D:45)>, 'AddressType': <dbus_fast.signature.Variant ('s', random)>, 'Alias': <dbus_fast.signature.Variant ('s', 20-D4-D9-6D-3D-45)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -66)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci1)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {6: <dbus_fast.signature.Variant ('ay', bytearray(b'\x01\t \x02~6\x9f\xa4\xaf\xec\xe7z\x07\xec(_\x86\xad>\xf2=\x89v\xd75#\xbb'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2024-10-23 20:09:27,939 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci1/dev_F9_82_74_04_59_8B', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', F9:82:74:04:59:8B)>, 'AddressType': <dbus_fast.signature.Variant ('s', public)>, 'Name': <dbus_fast.signature.Variant ('s', EB3A2420000402497)>, 'Alias': <dbus_fast.signature.Variant ('s', EB3A2420000402497)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -64)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', ['0000ff00-0000-1000-8000-00805f9b34fb'])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci1)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {19522: <dbus_fast.signature.Variant ('ay', bytearray(b'BLUETTF'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2024-10-23 20:09:27,943 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_20_D4_D9_6D_3D_45): ['org.bluez.Device1', {}, ['RSSI']]
2024-10-23 20:09:27,943 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved (/): ['/org/bluez/hci1/dev_20_D4_D9_6D_3D_45', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
2024-10-23 20:09:27,944 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_F9_82_74_04_59_8B): ['org.bluez.Device1', {}, ['RSSI']]
2024-10-23 20:09:27,944 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', False)>}, []]
2024-10-23 20:09:27,946 bleak.backends.bluezdbus.client MainThread DEBUG: Connecting to BlueZ path /org/bluez/hci1/dev_F9_82_74_04_59_8B
2024-10-23 20:09:28,136 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_F9_82_74_04_59_8B): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', True)>}, []]
2024-10-23 20:09:28,224 bleak.backends.bluezdbus.client MainThread DEBUG: retry due to le-connection-abort-by-local
2024-10-23 20:09:28,225 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_F9_82_74_04_59_8B): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', False)>}, []]
2024-10-23 20:09:28,225 bleak.backends.bluezdbus.client MainThread DEBUG: Device disconnected (/org/bluez/hci1/dev_F9_82_74_04_59_8B)
2024-10-23 20:09:28,225 bleak.backends.bluezdbus.client MainThread DEBUG: _cleanup_all(/org/bluez/hci1/dev_F9_82_74_04_59_8B)
[... cut ...]
    await client.start_notify(target_service, notification_handler, adapter="hci1")
  File "/home/<user>/bluetti/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 959, in start_notify
    assert_reply(reply)
  File "/home/<user>/bluetti/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownMethod] Method "StartNotify" with signature "" on interface "org.bluez.GattCharacteristic1" doesn't exist
```

dlech, 2024-10-23:
`BleakClientBlueZDBus` is for internal use only. `BleakClient` contains required code that is shared by all backends. You can still access the `BleakClientBlueZDBus` object from the `BleakClient` object via the `_backend` attribute.

See https://github.com/hbldh/bleak/blob/c98883bcfb08be62cdd7aa6eacefac52865c88c0/examples/mtu_size.py#L30

Furthermore, the `_acquire_mtu()` hack should not be needed with BlueZ version 5.72.

russellproud, 2024-10-24:
Thanks @dlech for the explanation and further details. I'll close this and go review from my side.

Cheers

closed 2024-10-24 by russellproud