hbldh/bleak #1678: How can I get raw advertisement data use bleak on windows 10

cqlhupt, 2024-10-27:
* bleak version: 0.22
* Python version: 3.10/3.12
* Operating System: Windows10 enterprise 21h1
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I have search more about how to get ble advertisement raw data on windows, and I found that on bleak v0.13  seem can do it，but on higher version some function and properties were changed. So I want to know how can I get ble advertisement raw data on latest bleak, I don't want parse advertisement because I want forward it to another device directly. Thank you for help me and have a good day.

### What I Did

search about how to get advertisement on windows
read bleak handbook on github

### Logs

NA

dlech, 2024-10-27:
Bleak has never been able to get raw advertisement data. 

Duplicate of #1296 and #540.

cqlhupt, 2024-10-30:
Thank you for your help, I have read source code of bleak backend, I think I can get advertisement like the backend, and process them directly.

closed 2024-10-30 by cqlhupt