hbldh/bleak #1679: Kivy example does not build

FilMarini, 2024-10-27:
* Python version: 3.10.12
* Operating System: Ubuntu 22.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

### Description

I tried to build the kivy example for Android. An error pops up not allowing to finish the build

### What I Did

Following the [README](https://github.com/hbldh/bleak/blob/develop/examples/kivy/README.md), I tried to build the kivy example.

This is what I did
```bash
pip3 install buildozer cython
buildozer android debug
```
And this is what I get (after a while)

```bash
[INFO]:    Building bleak for arm64-v8a
[INFO]:    bleak apparently isn't already in site-packages
[INFO]:    Installing bleak into site-packages
[INFO]:    -> directory context /home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/bleak/arm64-v8a__ndk_target_21/bleak
[DEBUG]:   -> running python3 setup.py install -O2 --root=/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/python-installs/bleakdemo/arm64-v8a --install-lib=.
[DEBUG]:   	Traceback (most recent call last):
[DEBUG]:   	  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/bleak/arm64-v8a__ndk_target_21/bleak/setup.py", line 6, in <module>
[DEBUG]:   	    setup(
[DEBUG]:   	  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/hostpython3/desktop/hostpython3/native-build/Lib/site-packages/setuptools/__init__.py", line 104, in setup
[DEBUG]:   	    return distutils.core.setup(**attrs)
[DEBUG]:   	           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[DEBUG]:   	  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/hostpython3/desktop/hostpython3/native-build/Lib/site-packages/setuptools/_distutils/core.py", line 147, in setup
[DEBUG]:   	    _setup_distribution = dist = klass(attrs)
[DEBUG]:   	                                 ^^^^^^^^^^^^
[DEBUG]:   	  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/hostpython3/desktop/hostpython3/native-build/Lib/site-packages/setuptools/dist.py", line 318, in __init__
[DEBUG]:   	    self.metadata.version = self._normalize_version(self.metadata.version)
[DEBUG]:   	                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[DEBUG]:   	  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/hostpython3/desktop/hostpython3/native-build/Lib/site-packages/setuptools/dist.py", line 354, in _normalize_version
[DEBUG]:   	    normalized = str(Version(version))
[DEBUG]:   	                     ^^^^^^^^^^^^^^^^
[DEBUG]:   	  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/hostpython3/desktop/hostpython3/native-build/Lib/site-packages/setuptools/_vendor/packaging/version.py", line 198, in __init__
[DEBUG]:   	    raise InvalidVersion(f"Invalid version: '{version}'")
[DEBUG]:   	setuptools.extern.packaging.version.InvalidVersion: Invalid version: 'bleak develop branch'
Exception in thread background thread for pid 245972:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/fmarini/.local/lib/python3.10/site-packages/sh.py", line 1634, in wrap
    fn(*rgs, **kwargs)
  File "/home/fmarini/.local/lib/python3.10/site-packages/sh.py", line 2636, in background_thread
    handle_exit_code(exit_code)
  File "/home/fmarini/.local/lib/python3.10/site-packages/sh.py", line 2327, in fn
    return self.command.handle_command_exit_code(exit_code)
  File "/home/fmarini/.local/lib/python3.10/site-packages/sh.py", line 821, in handle_command_exit_code
    raise exc
sh.ErrorReturnCode_1: 

  RAN: /home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/hostpython3/desktop/hostpython3/native-build/python3 setup.py install -O2 --root=/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/python-installs/bleakdemo/arm64-v8a --install-lib=.

  STDOUT:
Traceback (most recent call last):
  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/bleak/arm64-v8a__ndk_target_21/bleak/setup.py", line 6, in <module>
    setup(
  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/hostpython3/desktop/hostpython3/native-build/Lib/site-packages/setuptools/__init__.py", line 104, in setup
    return distutils.core.setup(**attrs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/fmarini/sink/rp-pico/prova/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/other_builds/hostpython3/desktop/hostpython3/native-build/Lib/site-packages/setuptools/_distutils/core.py", line 147, in setup
    _setup_distribution = dist = klass(attrs)
[... 126 lines cut ...]
#     XDG_SESSION_CLASS = 'user'
#     IBUS_DISABLE_SNOOPER = '1'
#     ZSH = '/home/fmarini/.oh-my-zsh'
#     DEFAULT_USER = 'fmarini'
#     PAGER = 'less'
#     LESS = '-R'
#     LSCOLORS = 'Gxfxcxdxbxegedabagacad'
#     LS_COLORS = 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:'
#     PS1 = '(env) %{%f%b%k%}$(build_prompt) '
#     VIRTUAL_ENV = '/home/fmarini/sink/rp-pico/prova/bleak/env'
#     VIRTUAL_ENV_PROMPT = '(env) '
#     PACKAGES_PATH = '/home/fmarini/.buildozer/android/packages'
#     ANDROIDSDK = '/home/fmarini/.buildozer/android/platform/android-sdk'
#     ANDROIDNDK = '/home/fmarini/.buildozer/android/platform/android-ndk-r25b'
#     ANDROIDAPI = '31'
#     ANDROIDMINAPI = '21'
# 
# Buildozer failed to execute the last command
# The error might be hidden in the log above this error
# Please read the full log, and search for it before
# raising an issue with buildozer itself.
# In case of a bug report, please add a full log with log_level = 2

```
I tried running it from both the `develop` branch as well as from the latest tag, `v0.22.3`

FilMarini, 2024-10-30:
The error is related to this line: https://github.com/hbldh/bleak/blob/develop/bleak/backends/p4android/recipes/bleak/__init__.py#L11

if you change it to `0.20.0` for example, buildozer finishes the build.

If I run the compiled example on my phone, the app crashes at start.
Reading the log, it seems that a significant error could be this one:
```
FATAL EXCEPTION: main
10-30 10:18:47.401 20466 20466 E AndroidRuntime: Process: com.example.bleuart, PID: 20466
10-30 10:18:47.401 20466 20466 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{com.example.bleuart/com.example.bleuart.MainActivity}: java.lang.IllegalStateException: You need to use a Theme.AppCompat theme (or descendant) with this activity.
```
Any idea how to solve this?
Thanks

Sahil-pixel, 2024-12-11:
Please solve this as soon as possible. We want this for kivy Android.