hbldh/bleak #1688: Typing: gatt descriptor obj

JPHutchins, 2024-11-13:
This interface contains an opaque Any type that can be improved with Generics.

https://github.com/hbldh/bleak/blob/e01e2640994b99066552b6161f84799712c396fa/bleak/backends/descriptor.py#L110-L111

dlech, 2025-05-10:
I'm not sure this would work since we also use `BleakGATTDescriptor` outside of the backends.

dlech, 2025-12-29:
I looked into this a little more. Each service/characteristic/descriptor type would require 3 generic args since they all reference each other, as would `BaseBleakClient`. And those 3 type args would have to be repeated everywhere we reference the types, which is a lot.

So this would make things way more verbose, harder to maintain and would be a breaking change on top of that (for type checking, not at runtime). So I think we are better off leaving things the way they are.

closed 2025-12-29 by dlech