hbldh/bleak #1689: [org.bluez.Error.InProgress] Operation already in progress

9acca9, 2024-11-18:
Im using the development Bleak version in Ubuntu 22.04

The thing is that i get this erro when try to run my code:
```
python mainBleak.py 
Connecting to device...
[org.bluez.Error.InProgress] Operation already in progress

```
It seems that Ubuntu is always scanning and well... Somebody here have experience with this issue? i try several things and nothing... 
ex.:
```
 bluetoothctl
Agent registered
[CHG] Controller 00:15:83:0C:BF:EB Pairable: yes
[bluetooth]# scan off
Failed to stop discovery: org.bluez.Error.Failed
[bluetooth]# 
```

Thanks

dlech, 2025-04-19:
I had this happen to me today. I have a USB Bluetooth dongle, so I unplugged it and plugged it back in and I stopped getting the "Operation already in progress" error when scanning. (Ubuntu 24.04, BlueZ 5.72).

So seems like a BlueZ issue.

hemangi825, 2025-06-12:
i get issue while write in BLE device, how to solve in nodejs i am using node-ble libs for that 
Please anyone suggest how to solve this issue