hbldh/bleak #1694: bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Not connected

Sahil-pixel, 2024-12-09:
* bleak version: 0.22.3
* Python version: 3.10
* Operating System: Ubuntu 22.04
* BlueZ version :  bluetoothctl: 5.64

### Description

I am trying to send some data to my BLE module . 
### What I Did

```python 
import asyncio
from bleak import BleakClient

# Replace these with your actual device address and characteristic UUID
DEVICE_ADDRESS = "8D:3C:82:19:5F:7C"  # Your BLE device's address
CHARACTERISTIC_UUID = "0000ffe1-0000-1000-8000-00805f9b34fb"  # UUID of the target BLE characteristic

async def main():
    async with BleakClient(DEVICE_ADDRESS) as client:
        if client.is_connected:
            print("Connected to device")
            # Send data (example: send [0x01, 0x02])
            data_to_send = bytearray([0x01, 0x02])
            await client.write_gatt_char(CHARACTERISTIC_UUID, data_to_send)
            print(f"Sent data: {data_to_send}")
        else:
            print("Failed to connect")

# Run the main loop
loop = asyncio.get_event_loop()
loop.run_until_complete(main())
```

### Logs
Connected to device
Traceback (most recent call last):
  File "/home/dixiong/data_sender.py", line 21, in <module>
    loop.run_until_complete(main())
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/dixiong/data_sender.py", line 14, in main
    await client.write_gatt_char(CHARACTERISTIC_UUID, data_to_send)
  File "/home/.local/lib/python3.10/site-packages/bleak/__init__.py", line 786, in write_gatt_char
    await self._backend.write_gatt_char(characteristic, data, response)
  File "/home/sahil/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 862, in write_gatt_char
    assert_reply(reply)
  File "/home/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Not connected

dlech, 2024-12-09:
Perhaps this issue? https://bleak.readthedocs.io/en/latest/troubleshooting.html#occasional-not-connected-errors-on-raspberry-pi

Sahil-pixel, 2024-12-10:
> Perhaps this issue? https://bleak.readthedocs.io/en/latest/troubleshooting.html#occasional-not-connected-errors-on-raspberry
This is not solving the issue.

dlech, 2024-12-19:
I suggest to log packets using wireshark as described in the troubleshooting page of the docs to see what is going on at a lower level.