hbldh/bleak #1697: partially unknown types in pyright

khusmann, 2024-12-15:
* bleak version: 0.22.1
* Python version: 3.10
* pylance version: v2024.7.1

### Description

In commands that make use of *kwargs, pyright complains that types are unknown

### What I Did

For example, on `BleakClient.connect()`

[image]

```
Type of "connect" is partially unknown
  Type of "connect" is "(**kwargs: Unknown) -> Coroutine[Any, Any, bool]"Pylance[reportUnknownMemberType](https://github.com/microsoft/pyright/blob/main/docs/configuration.md#reportUnknownMemberType)
```

Type errors are resolved if we change

```
async def connect(self, **kwargs) -> bool:
```

to

```
async def connect(self, **kwargs: Any) -> bool:
```

it looks like this should be done to all *args and **kwargs across all functions.

Or perhaps we should do something with `Unpack[]` like `BleakScanner` does for a few of its methods? https://github.com/hbldh/bleak/blob/e01e2640994b99066552b6161f84799712c396fa/bleak/__init__.py#L374

dlech, 2024-12-15:
Using `Unpack` sounds like the best option to me.

khusmann, 2024-12-15:
Yeah I wasn't familiar with `Unpack` until happened to see it here, but looks like the Right Way™  to go about this.

dlech, 2025-05-18:
Fixed in https://github.com/hbldh/bleak/commit/7da9c0ec42bf476ff37b7475eec3b83e8fb44a8c

closed 2025-05-18 by dlech

khusmann, 2025-05-18:
Awesome, thanks!