hbldh/bleak #1701: (Win11) Random error messages on every execution

shortcakesweets, 2024-12-17:
* bleak version: 0.22.3
* Python version: 3.8.10
* Operating System: Win11 22631.4602
* ~~BlueZ version (`bluetoothctl -v`) in case of Linux~~: No Linux Required

### Description

I'm trying to connect a remote controller to PC. The remote uses 'Just Connected' protocol.
I've built the project using python venv.
Bleak gives random error messages each time I execute it.
(It's strange because I've used bleak with no problem on this very same environment, but one day it stopped working)

Libraries I'm using on venv (checked with `pip list`):
```
async-timeout     4.0.3
bleak             0.22.3
bleak-winrt       1.2.0
pip               24.3.1
setuptools        56.0.0
typing_extensions 4.12.2
```

### What I Did

I used this code;

```python
import asyncio
import os
from bleak import BleakClient

address = "D0:1B:1F:82:11:23"

async def reproA():
    async with BleakClient(address) as client:
        print("Connected. Pairing...")
        await client.pair()
        print("Paired.")

if __name__ == '__main__':
    # clear previous logs
    os.system('cls' if os.name == 'nt' else 'clear')
    asyncio.run(reproA())
```

### Logs

**Error 1** : `asyncio.exceptions.CancelledError`
```
Traceback (most recent call last):
  File "repro.py", line 15, in <module>
    asyncio.run(reproA())
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
asyncio.exceptions.CancelledError
```

**Error 2** : `asyncio.exceptions.TimeoutError`
```
Traceback (most recent call last):
  File "repro.py", line 15, in <module>
    asyncio.run(reproA())
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "repro.py", line 8, in reproA
    async with BleakClient(address) as client:
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\__init__.py", line 570, in __aenter__
    await self.connect()
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\__init__.py", line 615, in connect   
    return await self._backend.connect(**kwargs)
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 487, in connect
    self.services = await self.get_services(
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\async_timeout\__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\async_timeout\__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

**Error 3** : `OSError: [WinError -2147483629] The object is closed`
```
Traceback (most recent call last):
  File "repro.py", line 15, in <module>
    asyncio.run(reproA())
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "repro.py", line 8, in reproA
    async with BleakClient(address) as client:
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\__init__.py", line 570, in __aenter__
    await self.connect()
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 487, in connect
    self.services = await self.get_services(
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 768, in get_services
    new_services.add_characteristic(
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\service.py", line 163, in add_characteristic
    self.__services[characteristic.service_handle].add_characteristic(
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\characteristic.py", line 92, in service_handle
    return int(self.obj.service.attribute_handle)
OSError: [WinError -2147483629] 개체가 닫혔습니다
```

**Error 4** : `OSError: [WinError -2147023673] The operation was canceled by the user`
```
Traceback (most recent call last):
  File "repro.py", line 15, in <module>
    asyncio.run(reproA())
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "repro.py", line 8, in reproA
    async with BleakClient(address) as client:
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\__init__.py", line 570, in __aenter__
    await self.connect()
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 487, in connect
    self.services = await self.get_services(
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 763, in get_services
    await FutureLike(characteristic.get_descriptors_async(*args)),
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 1129, in __await__
    yield self  # This tells Task to wait for completion.
OSError: [WinError -2147023673] 사용자가 작업을 취소했습니다.
```

**Error 5** : `OSError: [WinError -2147024809] The parameter is incorrect.`
```
Traceback (most recent call last):
  File "repro.py", line 15, in <module>
    asyncio.run(reproA())
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\dwjeong\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "repro.py", line 8, in reproA
    async with BleakClient(address) as client:
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\__init__.py", line 570, in __aenter__
    await self.connect()
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 487, in connect
    self.services = await self.get_services(
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 763, in get_services
    await FutureLike(characteristic.get_descriptors_async(*args)),
  File "C:\Users\dwjeong\Desktop\Shortcake\2.Projects\tool\btcli\venv\lib\site-packages\bleak\backends\winrt\client.py", line 1129, in __await__
    yield self  # This tells Task to wait for completion.
OSError: [WinError -2147024809] The parameter is incorrect.
```

dlech, 2024-12-17:
Sounds like it could be a not well-behaving Bluetooth adapter. Perhaps Intel integrated wireless/bluetooth?

You can see what is going on better at a low level by logging Bluetooth packets with Wireshark as described in the troubleshooting page of the docs.

shortcakesweets, 2024-12-20:
Hi, I've got the packet log using Wireshark.
I don't think the reason is from bluetooth malfunctioning. I've tested with several peripherals and server pairs, and all seems to fail.

The [log](https://github.com/user-attachments/files/18207400/ws_log_trim.zip) is from the environment above
- bleak version: 0.22.3
- Python version: 3.8.10
- Operating System: Win11 22631.4602

Meanwhile, I've found that using python 3.12 (where the dependencies are not bleak-winrt, but uses several packages) makes consistent error message. I'll try to log that in advance.

dlech, 2024-12-20:
I see a potential problem:

[image]

There is no response from the peripheral for this Read By Type Request. There should either be a Read By Type Response or an Error Response. 

It looks like the Bluetooth driver gives up after waiting 1/2 of a second for the response.

shortcakesweets, 2024-12-22:
If that's the case, I don't understand why directly connecting to the same computer-peripheral pair works properly but using bleak doesn't.

shortcakesweets, 2024-12-23:
Also if somebody sees this issue with Windows11 working environment, please give me
- python version
- used libraries (using `pip list`)
- minimal working code example

This would help a lot 👍