hbldh/bleak #1706: Unable to discover any device in MacOS Sequoia 15.2

nastiliano, 2025-01-05:
* bleak version: 0.22.3
* Python version: 3.13.1
* Operating System: MacOS Sequoia 15.2 (Intel Processor)

### Description

Unable to discover any device

### What I Did

Just with discover usage example:

    import asyncio
    from bleak import BleakScanner

    async def main():
        devices = await BleakScanner.discover()
        for d in devices:
            print(d)

    asyncio.run(main())

### Logs

Nothing appears in the conosole
No logs

Thank you very much and happy new year

dlech, 2025-01-05:
Was it working for you in older versions of macos? Are there any BLE peripherals that are advertising when you are scanning?

nastiliano, 2025-01-06:
Hello @dlech 

I don't know because I'm new in "Bleak" and I only tried "Bleak" in Ubuntu 24.10 - Linux 6.13.2 - x64 (it works fine) and MacOS Sequoia 15.2 (Intel processor).

No devices appear in MacOS.

This afternoon I'll try it in Windows Pro 11 x64 (completly updated).

Thank you very much.

nastiliano, 2025-01-06:
Confirmed.
In Windows Pro 11 x64 (completly updated) works completly fine!

Thank you very much.

anarion80, 2025-01-22:
Not sure if it helps, but I have the same issue when starting Home Assistant on MacOS 15.2, and here's the traceback:

```log
Fatal Python error: Aborted

Thread 0x000000017513f000 (most recent call first):
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 355 in wait
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/concurrent/futures/_base.py", line 451 in result
  File "/Users/anarion/Documents/Repos/tech-controllers/.devenv/state/venv/lib/python3.12/site-packages/homeassistant/components/recorder/core.py", line 665 in _wait_startup_or_shutdown
  File "/Users/anarion/Documents/Repos/tech-controllers/.devenv/state/venv/lib/python3.12/site-packages/homeassistant/components/recorder/core.py", line 753 in _run
  File "/Users/anarion/Documents/Repos/tech-controllers/.devenv/state/venv/lib/python3.12/site-packages/homeassistant/components/recorder/core.py", line 677 in run
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x0000000174133000 (most recent call first):
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/concurrent/futures/thread.py", line 90 in _worker
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1012 in run
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x0000000173127000 (most recent call first):
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/concurrent/futures/thread.py", line 90 in _worker
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1012 in run
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x000000017211b000 (most recent call first):
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/concurrent/futures/thread.py", line 90 in _worker
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1012 in run
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x000000017110f000 (most recent call first):
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/concurrent/futures/thread.py", line 90 in _worker
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1012 in run
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x0000000170103000 (most recent call first):
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/logging/handlers.py", line 1526 in dequeue
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/logging/handlers.py", line 1577 in _monitor
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1012 in run
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x000000016f0f7000 (most recent call first):
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/concurrent/futures/thread.py", line 90 in _worker
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1012 in run
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x00000001f83f8240 (most recent call first):
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 359 in wait
  File "/nix/store/qjrg3iljf6f6rs8sb5zlwx2zvv8i4cmm-python3-3.12.8-env/lib/python3.12/threading.py", line 655 in wait
  File "/Users/anarion/Documents/Repos/tech-controllers/.devenv/state/venv/lib/python3.12/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 85 in init
  File "/Users/anarion/Documents/Repos/tech-controllers/.devenv/state/venv/lib/python3.12/site-packages/bleak/backends/corebluetooth/scanner.py", line 77 in __init__
[... cut ...] v.video.plane, av.video.frame, av.video.stream, av.codec.codec, av.frame, av.audio.layout, av.audio.plane, av.audio.frame, av.audio.stream, av.audio.fifo, av.filter.pad, av.filter.link, av.filter.context, av.filter.graph, av.filter.filter, av.audio.resampler, psutil._psutil_osx, psutil._psutil_posix, objc._objc, CoreFoundation._CoreFoundation, CoreFoundation._inlines, Foundation._Foundation, Foundation._inlines, CoreBluetooth._CoreBluetooth, dispatch._dispatch, dispatch._inlines (total: 122)
```

nastiliano, 2025-01-22:
Hello,

I don't know, this is a job for "The Experts".

I can only say that "Bleak" works very well in my devices:
- Windows 11 Pro (x64)
- Ubuntu 24.10 Linux 6.13.2 (x64)

But not in my MacOS Sequoia 15.2 (Intel Processor)

Thanks.

dlech, 2025-01-24:
> Fatal Python error: Aborted

This looks like a crash in Python itself. 

Unfortunately, I'm not able to reproduce the issue. I just updated my OS to 15.2 and tried both the official and homebrew versions of Python 3.13.1. Running `examples/discover.py` works as expected for me with no crash.

nastiliano, 2025-01-29:
Hello.

Now, I try it again in my recently updated MacOS 15.3 device and it works fine.

Thanks

closed 2025-01-29 by nastiliano