hbldh/bleak #1712: Disconnected Reason 19

neil-brik, 2025-01-17:
Hello,

  I've got a Nordic nrf52dk running peripheral_uart and I'm using the uart_service.py example to try and connect to it.  However this doesn't connect and the RTT Viewer gives the error below repeated:

`00> [21:24:23.761,810] <inf> brikwiz_base: Connected D4:D8:53:4C:5D:23 (public)
00> [21:24:24.115,875] <inf> brikwiz_base: Disconnected: D4:D8:53:4C:5D:23 (public) (reason 19)` 

  Interestingly this also happens when you try to connect with the windows bluetooth console.  Is there a way to understand what is happening here as it'll be much easier if I can communicate to the device with a windows machine.

  Thanks.

Cheers,

Neil

dlech, 2025-01-17:
The best way to see what is going on at a lower level is to log Bluetooth packets with Wireshark as described in the troubleshooting page of the docs.

neil-brik, 2025-01-17:
Hello,

  Thanks, I'll give that a go on Monday morning.

Cheers,

Neil

neil-brik, 2025-01-20:
Hello,

  I tried this by following the instructions at https://bleak.readthedocs.io/en/latest/troubleshooting.html.  I installed wireshark and BTP.  However on executing the command 'c:\BTP\v1.14.0\x86\btvs.exe', wireshark starts and prompts me to do full packet connection or debug, whatever I click I get the  message from Wireshark 'Error reading from pipe: An existing connection was forcibly closed by the remote host.  (10054)'.  Google isn't showing anything to me on this as the error is quite generic.

  My setup is win 11 Pro, I've disconnected all the nordic devices from my computer and shut down J-Link RTT Viewer in case that is messing with something.

  Any ideas on how to proceed at all please?

Cheers,

Neil

neil-brik, 2025-01-20:
Hello,

  Thta's dine; looking at teh packet exchange we seem to be receiving the following:

1027	15.146039	controller	host	HCI_EVT	7	Rcvd Command Complete (LE Set Extended Scan Enable)
1028	15.379031	controller	host	HCI_EVT	7	Rcvd Encryption Change
1029	15.379126	host	controller	HCI_CMD	7	Sent Disconnect

  For some reason when we get encryption change this cause the windows computer to disconnect; I can't attach the full trace but it is available [here](https://brikwiz-my.sharepoint.com/:u:/p/neil_benn/Ea6OgNSE7_pOj5D0v7h7FwIBug8ltiwTWI7tv70WVDrRCQ?e=GaRT4l).  Do you have any ideas as to the reason it is disconnecting and can I avoid it?

  Has anyone else managed to get this example working in windows?

Cheers,

Neil

dlech, 2025-01-20:
Sounds just like this discussion: https://github.com/hbldh/bleak/discussions/1708