hbldh/bleak #1717: BleakScanner does not report device name when passively scanning

emece67, 2025-02-04:
Hi,

In fact I'm not sure if this is the expected behaviour or there's some issue when passively scanning for devices, but:
```
import asyncio
from bleak import BleakScanner

async def main():
    async with BleakScanner(scanning_mode='passive') as scanner:
        async for device, data in scanner.advertisement_data():
            print(device)

asyncio.run(main())
```
does not report a name to **any** device:
```
D8:A3:5C:87:39:05: None
5E:07:F5:F1:C5:DD: None
F4:B3:B1:82:F9:4C: None
70:F4:84:0B:03:78: None
A4:C1:38:B8:BE:70: None
```
But changing to active scanning `BleakScanner(scanning_mode='active')` does:
```
50:3E:4E:BB:2A:CC: S39 9414 LE
D8:A3:5C:87:39:05: None
A4:C1:38:6F:62:E1: None
A4:C1:38:6F:62:E1: M*68_T7_E1
5E:07:F5:F1:C5:DD: None
F4:B3:B1:82:F9:4C: SBBT-2Cf94c
```
I've seem some discussion here about `bleak` having problems in the past getting device names on Windows, but I do not have enough experience with BLE to know if this is a related problem or, simply, the expected behaviour. In case of the SBBT-2Cf94c device, I'm sure it is beaconing its shortened local name, and device M*68_T7_E1 is beaconing its complete local name.

I'm on Windows 11, Python 3.12.8, bleak 0.22.3.

Thanks & regards.

dlech, 2025-02-04:
By definition, passive scanning does not request a the SCAN_RSP from the peripheral device. So if the device name is only contained in SCAN_RSP, then it won't be possible to get that information. So it is entirely up to the peripheral if this should work or not.

emece67, 2025-02-05:
So I think this can be closed now.

Thanks a lot & regards.

closed 2025-02-05 by emece67