hbldh/bleak #1720: Parallels VM Crashes When Using Bluetooth on Windows 10

startupreboot, 2025-02-11:
Parallels VM Crashes When Using Bluetooth on Windows 10
Environment:

Host OS: macOS 13.7 Intel
Virtual Machine Software: Parallels Desktop 20
Guest OS: Windows 10 
Bluetooth Device: Raspberry PI

Issue Summary:
When attempting to connect a Bluetooth device or transfer data via BLE GATT inside a Windows 10 virtual machine running inside Parallels for Mac, the entire VM crashes unexpectedly.

Steps to Reproduce:

Start the Windows 10 virtual machine in Parallels.
Enable Bluetooth inside the VM.
Attempt to pair a Bluetooth device or transfer message over BLE GATT.
The Parallels VM crashes completely.

Code works perfectly inside the VM.

dlech, 2025-02-11:
Shouldn't you be reporting this to Parallels?

startupreboot, 2025-02-11:
It's difficult to determine if the issue is caused by the Bleak library or if it's a problem within Parallels itself. Other Bluetooth applications run without crashing the VM, so it seems specific to this use case. I will investigate further to narrow it down, and  I've reported it to them as well.

closed 2026-10-11 by github-actions[bot]