hbldh/bleak #1722: Bluetooth device closes almost immediately after connecting in pulse audio

Nosugrof, 2025-02-18:
I just ran into this issue on Ubuntu 22.04 with a bluetooth connection failing almost immediately after opening due to missing the "ServicesResolved=yes" setting. I'm using Pixel Buds Pro 2 ear buds and they work fine when I connect in bluetoothctl. 

To reproduce the issue first start vlc or another audio playback tool playing a long audio track. Next disconnect the era buds if they're currently connected. Then take the ear buds out of the case to initiate an auto-connect. The auto-detect in pulse audio makes the connection and it works for ~3 seconds and then drops. If I then open bluetoothctl I can see the device shows connected but it's not working and I don't see it in pavucontrol. If I run a connect command in bluetoothctl I get "[CHG] Device 5C:33:7B:6C:E9:49 ServicesResolved: yes" and then it works without dropping out.

I created a work around for this issue by adding a udev rules file that keys on the add event for the bluetooth device that invokes a script to call "bluetoothctl connect <device>" which corrects the issue. This way every time the bluetooth device is added it automatically calls connect on bluetoothctl to restore the "ServicesResolved=yes" setting. I had to include an 8 second delay in the script to allow time for the connect to fail before calling connect in bluetoothctl. 

Credit to "Anson VanDoren" for the udev hook framework.

Note that I'm using xfce and I have snap disabled.

## tail -f /var/log/syslog
Feb 17 21:11:11 latitude pulseaudio[1039]: Battery Level: 100%
Feb 17 21:11:11 latitude kernel: [13591.407249] input: Pixel Buds Pro 2 (AVRCP) as /devices/virtual/input/input50
Feb 17 21:11:11 latitude bluetoothd[850]: /org/bluez/hci0/dev_5C_33_7B_6C_E9_49/sep1/fd29: fd(27) ready
Feb 17 21:11:13 latitude root: Updated bluez_card.5C_33_7B_6C_E9_49 profile to High Fidelity Playback (A2DP Sink) profile
Feb 17 21:11:13 latitude bluetoothd[850]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Feb 17 21:11:13 latitude kernel: [13593.474582] hid-generic 0005:00E0:3005.0014: unknown main item tag 0x0
Feb 17 21:11:13 latitude kernel: [13593.474665] hid-generic 0005:00E0:3005.0014: hidraw0: BLUETOOTH HID v1.00 Device [Pixel Buds Pro 2] on 80:00:0b:da:7f:a4
Feb 17 21:11:17 latitude root: Issued connect on Pixel Buds Pro 2 : 5C:33:7B:6C:E9:49
Feb 17 21:11:19 latitude acpid: input device has been disconnected, fd 22
Feb 17 21:11:20 latitude pulseaudio[1039]: Battery Level: 100%
Feb 17 21:11:21 latitude root: Issued second connect on Pixel Buds Pro 2 : 5C:33:7B:6C:E9:49
Feb 17 21:11:21 latitude kernel: [13601.572367] input: Pixel Buds Pro 2 (AVRCP) as /devices/virtual/input/input51
Feb 17 21:11:23 latitude bluetoothd[850]: /org/bluez/hci0/dev_5C_33_7B_6C_E9_49/sep1/fd30: fd(27) ready
Feb 17 21:11:23 latitude root: Updated bluez_card.5C_33_7B_6C_E9_49 profile to High Fidelity Playback (A2DP Sink) profile
Feb 17 21:11:27 latitude root: Issued connect on Pixel Buds Pro 2 : 5C:33:7B:6C:E9:49

## udevadm monitor
KERNEL[13727.179865] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/bluetooth/hci0/hci0:35 (bluetooth)
UDEV  [13727.182199] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/bluetooth/hci0/hci0:35 (bluetooth)
KERNEL[13727.526831] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/bluetooth/hci0/hci0:35/0005:00E0:3005.0015 (hid)
KERNEL[13727.527027] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/bluetooth/hci0/hci0:35/0005:00E0:3005.0015/hidraw/hidraw0 (hidraw)
KERNEL[13727.527070] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/bluetooth/hci0/hci0:35/0005:00E0:3005.0015 (hid)
UDEV  [13727.528734] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/bluetooth/hci0/hci0:35/0005:00E0:3005.0015 (hid)
UDEV  [13727.531805] add      /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/bluetooth/hci0/hci0:35/0005:00E0:3005.0015/hidraw/hidraw0 (hidraw)
UDEV  [13727.532995] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/bluetooth/hci0/hci0:35/0005:00E0:3005.0015 (hid)
KERNEL[13729.088302] add      /devices/virtual/input/input52 (input)
KERNEL[13729.088525] add      /devices/virtual/input/input52/event20 (input)
KERNEL[13737.335233] remove   /devices/virtual/input/input52/event20 (input)
KERNEL[13737.341360] remove   /devices/virtual/input/input52 (input)
UDEV  [13739.267716] add      /devices/virtual/input/input52 (input)
UDEV  [13739.269206] add      /devices/virtual/input/input52/event20 (input)
UDEV  [13739.269936] remove   /devices/virtual/input/input52/event20 (input)
UDEV  [13739.270671] remove   /devices/virtual/input/input52 (input)

cat 50-pixel-buds-pro-2.rules
ACTION=="add", SUBSYSTEM=="input", ATTR{id/vendor}=="00e0", ATTR{id/product}=="3005", RUN+="/home/doug/bin/setearbudprofile.sh"

cat setearbudprofile.sh
#!/bin/bash

sleep 2

# run "pactl list" to see the profile names.

sudo -u '#1000' XDG_RUNTIME_DIR=/run/user/1000 \
    pactl set-card-profile bluez_card.5C_33_7B_6C_E9_49 a2dp_sink

logger "Updated bluez_card.5C_33_7B_6C_E9_49 profile to High Fidelity Playback (A2DP Sink) profile"

## Issue a connect to force ServicesResolved = yes
sudo -u '#1000' bluetoothctl connect 5C:33:7B:6C:E9:49

sleep 4
logger "Issued connect on Pixel Buds Pro 2 : 5C:33:7B:6C:E9:49"

sleep 4
sudo -u '#1000' bluetoothctl connect 5C:33:7B:6C:E9:49
logger "Issued second connect on Pixel Buds Pro 2 : 5C:33:7B:6C:E9:49"

## Notes: 
## handsfree_head_unit is the profile name for HFP with mSBC codec

pactl list (last card only)
Card #23
        Name: bluez_card.5C_33_7B_6C_E9_49
        Driver: module-bluez5-device.c
        Owner Module: 47
        Properties:
                device.description = "Pixel Buds Pro 2"
                device.string = "5C:33:7B:6C:E9:49"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headset"
                bluez.path = "/org/bluez/hci0/dev_5C_33_7B_6C_E9_49"
                bluez.class = "0x244404"
                bluez.alias = "Pixel Buds Pro 2"
                device.icon_name = "audio-headset-bluetooth"
                device.intended_roles = "phone"
                bluetooth.battery = "100%"
                bluetooth.codec = "sbc"
        Profiles:
                a2dp_sink: High Fidelity Playback (A2DP Sink) (sinks: 1, sources: 0, priority: 40, available: yes)
                handsfree_head_unit: Handsfree Head Unit (HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_sink
        Ports:
                headset-output: Headset (type: Headset, priority: 0, latency offset: 0 usec, availability unknown)
                        Part of profile(s): a2dp_sink, handsfree_head_unit
                headset-input: Headset (type: Headset, priority: 0, latency offset: 0 usec, availability unknown)
                        Part of profile(s): handsfree_head_unit

dlech, 2025-02-18:
Bleak is a Python library for using Bluetooth low energy. This seems related to Bluetooth audio, so this doesn't seem like the right place to be reporting the issue. If the issue is still a problem with the latest BlueZ, then I would suggest reporting to https://github.com/bluez/bluez. Or if it has been fixed in BlueZ, then report the issue to Ubuntu to backport the fix.

closed 2025-02-18 by dlech

Nosugrof, 2025-02-18:
Thanks for the guidance. I just opened an issue in the bluez repo as suggested. Agree that's the correct place for this issue.