hbldh/bleak #1729: The scanning example in break cannot search for nrf52840 extended Bluetooth broadcast

JufeiFan, 2025-03-26:
windows 10

[image]

Using the scanning example in the break project, the Bluetooth extension broadcast of nrf52840 cannot be scanned.

dlech, 2025-03-26:
Duplicate of #1347?

JufeiFan, 2025-03-27:
> Duplicate of [#1347](https://github.com/hbldh/bleak/issues/1347)?

This is the latest issue I have encountered so far, not a replica of other problems. Does the break library currently support scanning and extending Bluetooth broadcasting.

JufeiFan, 2025-03-27:
At present, I want to use the break library to connect to the nrf52840 extended broadcast low-power Bluetooth on Windows. Does the library currently support it? Please give a clear answer. Thank you.

JufeiFan, 2025-03-27:
Which version of break currently supports it

JufeiFan, 2025-03-28:
@dlech

JufeiFan, 2025-03-28:
@EdwardBetts

JufeiFan, 2025-03-28:
@adamse

brycewilkins, 2025-06-11:
@JufeiFan Its possible you are encountering the issue that extended advertising is disabled by default for some platforms (Windows).

See https://github.com/hbldh/bleak/pull/1705 for the issue and resolution.

closed 2025-08-09 by dlech