hbldh/bleak #1730: Implicit optional in BleakClient

patman15, 2025-03-29:
Hi!

I'm really a fan of the library and using it a lot, so I ran into this issue:
Would it be possible (I'm volunteering to do a PR) to change the function signature of 

https://github.com/hbldh/bleak/blob/e01e2640994b99066552b6161f84799712c396fa/bleak/__init__.py#L723-L728

to

```python
    async def write_gatt_char(
        self,
        char_specifier: Union[BleakGATTCharacteristic, int, str, uuid.UUID],
        data: Buffer,
        response: bool | None = None,
    ) -> None:
```

to avoid mypy error
```
PEP 484 prohibits implicit Optional. Accordingly, mypy has changed its default to no_implicit_optional=True
Use https://github.com/hauntsaninja/no_implicit_optional to automatically upgrade your codebase
```
I have a number of classes derived from `BleakClient` and I get those complaints all over the place. I'm not sure there is any reason it was done like this or just missed out. Thanks!

dlech, 2025-03-29:
Hmm... We really want people to be explicit here and always provide this argument. But we have to leave it optional for backwards compatibility.

I'm OK with doing as you suggest. But we will need to use `Optional[bool]` since we still support Python < 3.10.

And we could consider doing `warn("Omitting the response argument is not supported and may be allowed in the future", DeprecationWarning, stacklevel=2)` while we are touching this.

patman15, 2025-03-29:
Sounds good to me, if you want me to support, how can I help?

dlech, 2025-03-29:
Happy to take a PR for this.

PR #1731 (2025-04-05): Fix BleakClient response type
fixes #1730
closed 2025-04-05 by dlech