hbldh/bleak #1735: Python scripts relying on bleak stops working after putting mac to sleep and waking up again

sclsj, 2025-04-06:
System: macOS 12.6.7, m1 max macbook 16''
Bleak 0.22.3

Example program 1: script written by chatGPT to record bluetooth thermostat, used with python 3.10.12

[goveedb.py.zip](https://github.com/user-attachments/files/19619125/goveedb.py.zip)

Problem: new readings stops when I wake up mac. Have to kill program and rerun to start getting readings again

Example program 2: homeassistant 2025.3.4, used with python 3.13.1

Problem 1: after adding sonicare integration and govee integration, new readings and updates stopped after waking up from sleep. Have to restart HA to fix. During shutdown (either restart via web interface or control+c in terminal), python crashes: 

```
Crashed Thread:        1  Dispatch queue: bleak.corebluetooth

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x00000001bb6c1980
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [98878]

Application Specific Information:
BUG IN CLIENT OF LIBPTHREAD: pthread_exit() called from a thread not created by pthread_create()

Thread 0::  Dispatch queue: com.apple.main-thread
0   Python                        	       0x1010fbcac subtype_traverse + 340
1   Python                        	       0x1011d6fc0 deduce_unreachable + 228
2   Python                        	       0x1011d5208 gc_collect_main + 408
3   Python                        	       0x10120a220 interpreter_clear + 540
4   Python                        	       0x1012069d8 finalize_interp_clear + 76
5   Python                        	       0x10120524c _Py_Finalize + 380
6   Python                        	       0x1012397c4 Py_RunMain + 596
7   Python                        	       0x101239ecc pymain_main + 352
8   Python                        	       0x101239fe0 Py_BytesMain + 40
9   dyld                          	       0x1009e108c start + 520

Thread 1 Crashed::  Dispatch queue: bleak.corebluetooth
0   libsystem_pthread.dylib       	       0x1bb6c1980 pthread_exit + 116
1   Python                        	       0x101225f98 PyThread_exit_thread + 40
2   Python                        	       0x1011dd688 take_gil + 544
3   Python                        	       0x10120cb0c PyGILState_Ensure + 88
4   _objc.cpython-313-darwin.so   	       0x12f1491f4 method_stub + 240
5   libffi.dylib                  	       0x1c9960f34 ffi_closure_SYSV_inner + 816
6   libffi.dylib                  	       0x1c99581e8 ffi_closure_SYSV + 56
7   CoreBluetooth                 	       0x1cd4c8c4c -[CBCentralManager handlePeripheralDiscovered:] + 1496
8   CoreBluetooth                 	       0x1cd4cab94 -[CBCentralManager handleMsg:args:] + 284
9   CoreBluetooth                 	       0x1cd50baac -[CBManager xpcConnectionDidReceiveMsg:args:] + 208
10  CoreBluetooth                 	       0x1cd4fafdc __30-[CBXpcConnection _handleMsg:]_block_invoke + 68
11  libdispatch.dylib             	       0x1bb4fe5f0 _dispatch_call_block_and_release + 32
12  libdispatch.dylib             	       0x1bb5001b4 _dispatch_client_callout + 20
13  libdispatch.dylib             	       0x1bb5078a8 _dispatch_lane_serial_drain + 668
14  libdispatch.dylib             	       0x1bb508438 _dispatch_lane_invoke + 444
15  libdispatch.dylib             	       0x1bb507784 _dispatch_lane_serial_drain + 376
16  libdispatch.dylib             	       0x1bb508404 _dispatch_lane_invoke + 392
17  libdispatch.dylib             	       0x1bb512c98 _dispatch_workloop_worker_thread + 648
18  libsystem_pthread.dylib       	       0x1bb6c0360 _pthread_wqthread + 288
19  libsystem_pthread.dylib       	       0x1bb6bf080 start_wqthread + 8
```

sclsj, 2025-04-06:
[HA crash while quit.zip](https://github.com/user-attachments/files/19619138/HA.crash.while.quit.zip)
Crash reports for second program

dlech, 2025-05-10:
> macOS 12.6.7

I don't have such an old version to test with. I suspect this could be fixed in newer versions of macOS.

I tried running the detection callback example and putting my mac to sleep (macOS 15.4.1) and waking it back up and I don't get a crash.