hbldh/bleak #1737: `BleakClient` fails to track address updates for devices using RPA on Linux with BlueZ.

vhakulinen, 2025-04-19:
#### Summary
When pairing with a device that uses Resolvable Private Addresses (RPA), `BleakClient` does not update its internal `address` to reflect the new address assigned by BlueZ. This leads to inconsistencies when interacting with the device post-pairing.

#### Steps to Reproduce
Run the code below:

```python
import asyncio
import subprocess

import bleak

class MyBleakClient(bleak.BleakClient):
    async def get_bluez_address(self) -> str:
        from dbus_fast.message import Message
        from bleak.backends.bluezdbus import defs

        m = Message(
            destination=defs.BLUEZ_SERVICE,
            path=await self._backend._get_device_path(),
            interface=defs.PROPERTIES_INTERFACE,
            member="Get",
            signature="ss",
            body=[defs.DEVICE_INTERFACE, "Address"],
        )
        reply = await self._backend._bus.call(m)

        return reply.body[0].value

async def main():
    original_addr = "75:e5:ed:87:1c:28"
    d = await bleak.BleakScanner.find_device_by_address(original_addr)
    assert d is not None
    c = MyBleakClient(d)

    await c.connect()
    print("address before pairing", c.address)
    print("bluez address before pairing", await c.get_bluez_address())

    await c.pair()
    print("address after pairing", c.address)
    bluez_addr = await c.get_bluez_address()
    print("bluez address after pairing", bluez_addr)

    await c.disconnect()

    print(
        "scan for original",
        await bleak.BleakScanner.find_device_by_address(original_addr),
    )
    print("scan for bluez", await bleak.BleakScanner.find_device_by_address(bluez_addr))

if __name__ == "__main__":
    # Run bluetoothctl in the background as the pairing agent.
    with subprocess.Popen(
        ["bluetoothctl"],
        stdout=subprocess.PIPE,
        stderr=subprocess.PIPE,
        stdin=subprocess.PIPE,
    ):
        asyncio.run(main())

```

#### Expected Behavior
`BleakClient.address` updates to match the new address assigned by BlueZ after pairing.

#### Actual Behavior
The output shows `BleakClient.address` remains unchanged, while BlueZ updates the address:

```
address before pairing 75:E5:ED:87:1C:28
bluez address before pairing 75:E5:ED:87:1C:28
address after pairing 75:E5:ED:87:1C:28
bluez address after pairing EA:28:4F:29:B1:40
scan for original None
scan for bluez EA:28:4F:29:B1:40: xxxxx
```

#### Root Cause
`BleakClient` does not listen for D-Bus property changes (`org.freedesktop.DBus.Properties.PropertiesChanged`) to update the `address` property when BlueZ updates it.

I have experience with BLE on Android and iOS, and neither of these platforms change the address under your feet. I'm not sure if this root cause is because of a incorrect behavior of bluez or if its working as designed. Either way, I think this should be (a) fixed in bleak, or (b) mentioned in the documentation and a workaround added to the bluez backend.

#### Impact
- Device scanning fails with the old address post-pairing.
- Applications relying on `address` for identification experience inconsistencies.

#### Proposed Fix
Listen for `PropertiesChanged` signals on the `org.bluez.Device1.Address` property and update `BleakClient.address` accordingly.

#### Environment
- **OS**: Pop-os 22.04
- **BlueZ**: 5.64
- **Bleak**: 0.22.2
- **Python**: 3.10

dlech, 2025-04-19:
Thanks for the detailed explanation, this is an interesting problem.

It sounds like the address only changes when pairing, so we could probably get away with updating the address at that time rather than having to have an address change event that users subscribe to to be notified of changes.

Also, we should be able to use the BlueZManager that is running the in background to get the updated address since it is listening for all property changes rather than having to do an async property get call.

I would also like to see how this works on macOS and Windows. Since macOS doesn't give the address, but rather a UUID, I have a feeling that it internally handles this address mapping, so the UUID never changes. I couldn't find Windows docs that gave any hints on how it would handle this.

vhakulinen, 2025-04-19:
I wonder if there is some kind of unique ID for each device in bluez that is comparable to the macOS/iOS UUIDs? The dbus path to the device keeps reflecting the original address while the _actual_ address is changed after pairing, i.e. dbus path is `/org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX` while the address is `YY_YY_YY_YY_YY_YY`.

dlech, 2025-04-19:
> I wonder if there is some kind of unique ID for each device in bluez that is comparable to the macOS/iOS UUIDs

In the case of a paired device that uses random private address, yes, I think so. It just makes it look like a Bluetooth address to reuse the existing property.

https://github.com/bluez/bluez/blob/master/doc/org.bluez.Device.rst#string-addresstype-readonly says:

>  If remote device uses privacy than [sic] before pairing this represents address type used for connection and Identity Address after pairing.

So the "Identity Address" would be the new address you see after pairing.

vhakulinen, 2025-04-20:
Additional observation: after pairing, sometimes `read_gatt_char` wont resolve until I disconnect the device and reconnect with the new "identity address". This is something I just ran into, and haven't seen this on other than one device...

dlech, 2025-04-20:
I wonder if #1741 would help with that. It essentially allows pairing before connecting. (We would still need to add a bit to update the `address` attribute to the new identity address.)