hbldh/bleak #1743: write_gatt_char default behavior

patman15, 2025-05-05:
Hi!

Sorry, I know, I started the discussion in #1730 and it led to #1736. After both were done I started modifying my code to always explicitly give the argument. Unfortunately, I have a unified code for many devices so I don't know if the characteristics supports `write` or `write-without-response` and thus would like to stay with the (now deprecated) default behavior as best guess.
Should I now replicate https://github.com/hbldh/bleak/blob/fc8fa2ad3ea2d1e88321e184aa79bc686d2d862b/bleak/__init__.py#L794 meaning
```python
        if isinstance(char_specifier, BleakGATTCharacteristic):
            characteristic = char_specifier
        else:
            characteristic = self.services.get_characteristic(char_specifier)

        if not characteristic:
            raise BleakCharacteristicNotFoundError(char_specifier)

        if response is None:
            # if not specified, prefer write-with-response over write-without-
            # response if it is available since it is the more reliable write.
            response = "write" in characteristic.properties
```
outside the library and feed it back to `write_gatt_char()` ?
Sounds a bit strange to me and "a lot of" code for bringing the *best-guess-behaviour* back.
Again, sorry, for not thinking about this before, when trying to adhere to the *make it explicit* suggestion in the docs.
@brianrho FYI

brianrho, 2025-05-05:
Yeah, pretty much. This may be more code than you had to write before, but it's important to remember that correct behaviour doesn't promise smaller code -- smaller code is after all often achieved by making a lot of unhealthy assumptions.

It should also be straightforward for you to wrap all this in a function of your own, or sub-class `BleakClient` to create your own custom method, say, `write_gatt_char_i()` that use a best-guess automatically -- and use this throughout your application.

dlech, 2025-05-05:
> It's also worth remembering that Python's argument for explicitness is a high-level principle ultimately advocating for code clarity, and not necessarily arguing that default arguments should go away

I wrote that in the docs to explain things in a way that Python programmers would understand. 😉  But the real issue is deeper. There is just too much broken hardware out there that the best guess option isn't really doing any favors for users that find themselves getting an error when it doesn't work and they don't know very much about how BLE works. So being explicit here really about intentionally selecting what the Bluetooth hardware was designed to do rather than relying on a best guess.

If we could start over from scratch, I would just make this two different functions instead of one with an option. But too late to do that now without breaking the whole world.

Anyway, if this best guess is still useful in some cases, I'm OK with dropping the deprecation.

brianrho, 2025-05-05:
In other words, I'd recommend:

- From b268261, keep the `Optional[bool] = None` change to satisfy linters
- Remove all mentions of deprecating `response=None`. Let it be a perfectly natural default argument, and briefly explain how the best-guess works

If a user cares about `response` at all, they'll discover the `response` easily enough from the docs and use it. If they don't care, then Bleak behaves no worse than the user's application could do, by guessing. 

This holds even if the remove Server doesn't report its Characteristic properties correctly. It would be up to the user to debug and realize said Server has a bug and so manually supply the `response` argument, based off experiments or information obtained by other means.

brianrho, 2025-05-05:
> I wrote that in the docs to explain things in a way that Python programmers would understand

Ah, makes sense :)

patman15, 2025-05-06:
Hi! I would basically second what you wrote in your [message](https://github.com/hbldh/bleak/issues/1743#issuecomment-2852613555) @brianrho. If @dlech still wants to push users to think about what option they use and make them early aware of the issues with the wrong devices, we could also do `Optional[bool] | Literal['auto']` to also have to explicitly request automatic behavior. But documentation alone and defaulting to `None` is totally fine for me. As a user, I was triggered by the linters and then read the description and then got confused. 😉 

Btw. there is a [wrapper for connection issues](https://github.com/Bluetooth-Devices/bleak-retry-connector) wouldn't it make sense to make something similar for this issue with the devices? Basically, I'm now trying to set this up for my integration, but I could make a library out of it if it makes sense.

Does anyone of you have some experience with a good strategy to find the right way? E.g.,
  1. Try with `write` if available in properties, else or if `write` fails
  2. try `write-without-response`

I'm looking for a strategy that covers most devices somehow. My device does not report `write` but it still works.
I would prefer `write` over `write-without-response` but I'm not sure how to distinguish a connection issue from the wrong operation. E.g., will `write` always cause an exception if it is the wrong method?

dlech, 2025-05-06:
> E.g., will `write` always cause an exception if it is the wrong method?

It entirely depends on how well the Bluetooth peripheral developers did their job. I've seen peripherals that just crash if you do write instead of write-without-repose even though the characteristic property says that it supports both.

For devices that implement well-known profiles, we should know from the profile if write or write-without-response is required (or in some cases optional). For custom characteristics that don't have documentation, reverse-engineering should be done to see that the manufacture's app does to know what the "right way" to use a characteristic is.

I don't think trying to bake something into the library like this seems like a good idea since it could cause just as many problems as it solves. It will be better if people do a little homework to figure out what the right option is.

However, I've been thinking about adding a CLI option to Bleak that would let users poke at a BLE device to be able to try different things out. So adding some kind of "which kind of write do I need?" helper to that could be useful.

brianrho, 2025-05-07:
I agree with @dlech -- that much guesswork has no place in a library that is backed by specification, and intends to be as generic as possible (and thus mustn't be too opinionated).

If you identify non-compliant Servers, you'll need to debug and workaround the issue(s), on a case-by-case basis -- there's no general solution for weirdness. Research, reverse-engineer, etc. and then use your results to supply the value of `response`.

One problem with your proposed strat is that it's possible for the `"write" in properties` check to return `False`, even though the device _does_ support `write` and not `write-with-response`. A failed `write-without-response` inherently won't return any errors from the remote device -- the remote's baseband would just ACK the LL packet and it would get discarded somewhere in the Host. Or the device would just crash like @dlech said, if it's reaaally badly written.

Personally, I've not used a lot of random Peripherals, especially non-compliant ones, so I don't have much advice here. I'd personally just hard-code the `response` once I'd figured out what this bad device truly supports, and call it a day (there can't that many such products, surely?). Perhaps this can be included in the examples (for a known non-compliant Server) to show another usecase where one might want to provide an explicit `response` argument.

patman15, 2025-05-08:
Ok, all clear. I meanwhile found a solution for me. I'm using the same logic as the library, but actively set it. Might revert that later or enhance depending on devices, that are faulty. Thanks a lot.
Let me know if you want me to help with a PR according to the suggestion in https://github.com/hbldh/bleak/issues/1743#issuecomment-2852613555

dlech, 2025-05-08:
> Let me know if you want me to help with a PR according to the suggestion in [#1743 (comment)](https://github.com/hbldh/bleak/issues/1743#issuecomment-2852613555)

Would be great if you could make a PR to removed the deprecation warning and fix up the docs.

closed 2025-05-16 by dlech

patman15, 2025-05-16:
Sorry didn't manage this week. :-(

dlech, 2025-05-16:
No worries. I actually had some extra time and energy to work on this for once. 😄