hbldh/bleak #1747: BlueZ: devices visible in btmon but not Bleak in default scanning_mode

alexandervlpl, 2025-05-14:
### Setup:

- Bleak 0.22.3
- Raspberry Pi OS Bookworm 2024-11-19
- Raspberry Pi 5

### Issue:

Running examples/discover.py does not find my BLE device (ESP32), even though it shows up in `bluetoothctl` and `btmon`:

```
> HCI Event: LE Meta Event (0x3e) plen 42                                                    #159 [hci0] 241.402582
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 94:54:C5:A8:14:7A (Espressif Inc.)
        Data length: 30
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        TX power: 9 dBm
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific
        Peripheral Conn. Interval: 0x0006 - 0x0010
        RSSI: -75 dBm (0xb5)
@ MGMT Event: ADV Monitor Device Found (0x002f) plen 46                                  {0x0001} [hci0] 241.402590
        Handle: 0
        Address: 94:54:C5:A8:14:7A (Espressif Inc.)
        Addr Type: 1
        RSSI: -75
        Flags: 0x00000000
        AD Data Len: 30
        AD Data: 020106020a091107b4df5a1c3f6bf4bfea4a820304901a02051206001000
```

### Workaround:

What does work is explicitly scanning in passive mode with patterns that accept all devices:

```
from bleak.backends.bluezdbus.scanner import BlueZScannerArgs
from bleak.backends.bluezdbus.advertisement_monitor import OrPattern
from bleak.assigned_numbers import AdvertisementDataType

PASSIVE_SCANNER_ARGS = BlueZScannerArgs(
    or_patterns=[
        OrPattern(0, AdvertisementDataType.FLAGS, b"\x06"),
        OrPattern(0, AdvertisementDataType.FLAGS, b"\x1a"),
    ]
)

devices = await BleakScanner.discover(
    return_adv=True,
    scanning_mode="passive",
    bluez=PASSIVE_SCANNER_ARGS
)
```

dlech, 2025-05-14:
That's odd. If you enable logging (e.g. set environment variable `BLEAK_LOGGING=1`), are there D-Bus property change messages in the not working case?

If yes, then this would be a Bleak bug. If no, then this would be a BlueZ bug.

alexandervlpl, 2025-05-15:
It turns out this strange state is caused by running a 2.4 GHz WiFi hotspot and active Bluetooth at the same time using the onboard wireless on the Pi. Only passive scan works, I can't get device names or connect. As soon as I kill WiFi everything returns to normal.

I'm guessing `bluetoothctl` and `btmon` do passive scans, so that's why I was seeing devices there but not in Bleak.

Hardware issue, but may be worth adding a note on interference in Troubleshooting?

closed 2025-05-16 by dlech