hbldh/bleak #1749: MTU exchange inconsistent

SimonJonsson1999, 2025-05-14:
I am trying to connect to a peripheral and read from some services that need bonding. When i connect I sometimes get a MTU exchange and sometimes not. 

Running the same code yields different results.
When i can connect successfully:
DEBUG:bleak.backends.winrt.scanner:17 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
Scanning stop, target device 70:54:64:D7:C5:D4
DEBUG:bleak.backends.winrt.client:Connecting to BLE device @ 70:54:64:D7:C5:D4
DEBUG:bleak.backends.winrt.client:getting services (service_cache_mode=None, cache_mode=None)...
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: BluetoothLE#BluetoothLE68:c6:ac:54:96:23-70:54:64:d7:c5:d4, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>
DEBUG:bleak.backends.winrt.client:max_pdu_size_changed_handler: 247
DEBUG:bleak.backends.winrt.client:70:54:64:D7:C5:D4: services changed
DEBUG:bleak.backends.winrt.client:70:54:64:D7:C5:D4: services changed
DEBUG:bleak.backends.winrt.client:70:54:64:D7:C5:D4: services changed
DEBUG:bleak.backends.winrt.client:70:54:64:D7:C5:D4: services changed
DEBUG:bleak.backends.winrt.client:70:54:64:D7:C5:D4: services changed
DEBUG:bleak.backends.winrt.client:70:54:64:D7:C5:D4: services changed
Connected to target

and when i can not:

DEBUG:bleak.backends.winrt.client:Connecting to BLE device @ 70:54:64:D7:C5:D4
DEBUG:bleak.backends.winrt.scanner:17 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
DEBUG:bleak.backends.winrt.client:getting services (service_cache_mode=None, cache_mode=None)...
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: BluetoothLE#BluetoothLE68:c6:ac:54:96:23-70:54:64:d7:c5:d4, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>
Connected to target
DEBUG:bleak.backends.winrt.client:max_pdu_size_changed_handler: 247
DEBUG:bleak.backends.winrt.client:max_pdu_size_changed_handler: 23
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: BluetoothLE#BluetoothLE68:c6:ac:54:96:23-70:54:64:d7:c5:d4, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.CLOSED: 0>
DEBUG:bleak.backends.winrt.client:closing requester
DEBUG:bleak.backends.winrt.client:closing session

I assume it has to do with the max_pdu_size_changed_handler: 23. But i dont understand how to influence this. As i understand it this is negotiated by the ble driver inside my windows machine. Can someone help me or guide me where to look for more info?

Here are some wireshark ble sniffer logs of the same 2 connection attempts.

Not wroking:

[image]

Working:

[image]

SimonJonsson1999, 2025-05-14:
After looking closer at the first LL_FEATURE_REQ, the more data bit in the header is True when it works, and False when it don't work.
Not sure if this is the root cause but my suspicion is that it is related
[image]

dlech, 2025-05-14:
This is a know quirk of how Windows works with some devices.

Did you follow the suggested workaround in the docs to wait for the updated MTU value?

https://bleak.readthedocs.io/en/latest/api/index.html#bleak.backends.characteristic.BleakGATTCharacteristic.max_write_without_response_size

SimonJonsson1999, 2025-05-15:
> This is a know quirk of how Windows works with some devices.
> 
> Did you follow the suggested workaround in the docs to wait for the updated MTU value?
> 
> https://bleak.readthedocs.io/en/latest/api/index.html#bleak.backends.characteristic.BleakGATTCharacteristic.max_write_without_response_size

I tried like this:
`    async def _connect(self, device):
        self.client = BleakClient(device)
        await self.client.connect()
        print("Connected to target")
        
        CHAR_UUID = "00002a00-0000-1000-8000-00805f9b34fb"
        char = self.client.services.get_characteristic(CHAR_UUID)
        async with asyncio.timeout(10):
            while char.max_write_without_response_size == 20:
                print(char.max_write_without_response_size)
                await asyncio.sleep(0.5)`
but i don't quite get it to work:

DEBUG:bleak.backends.winrt.client:Connecting to BLE device @ 70:54:64:D7:C5:D4
DEBUG:bleak.backends.winrt.client:getting services (service_cache_mode=None, cache_mode=None)...
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: BluetoothLE#BluetoothLE68:c6:ac:54:96:23-70:54:64:d7:c5:d4, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>
Connected to target
20
DEBUG:bleak.backends.winrt.client:max_pdu_size_changed_handler: 247
DEBUG:bleak.backends.winrt.client:max_pdu_size_changed_handler: 23
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: BluetoothLE#BluetoothLE68:c6:ac:54:96:23-70:54:64:d7:c5:d4, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.CLOSED: 0>
DEBUG:bleak.backends.winrt.client:closing requester
DEBUG:bleak.backends.winrt.client:closing session
Traceback (most recent call last):
  File "C:\Users\8957\Documents\tests\BLEDeviceManager\BLEAKDeviceManager.py", line 144, in <module>
    asyncio.run(main())
  File "C:\Users\8957\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\8957\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\8957\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 650, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\8957\Documents\tests\BLEDeviceManager\BLEAKDeviceManager.py", line 132, in main
    await ble_device.initilize()
  File "C:\Users\8957\Documents\tests\BLEDeviceManager\BLEAKDeviceManager.py", line 19, in initilize
    await self._connect(target_device)
  File "C:\Users\8957\Documents\tests\BLEDeviceManager\BLEAKDeviceManager.py", line 88, in _connect
    while char.max_write_without_response_size == 20:
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\8957\Documents\tests\venv\Lib\site-packages\bleak\backends\characteristic.py", line 112, in max_write_without_response_size
    return self._max_write_without_response_size()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\8957\Documents\tests\venv\Lib\site-packages\bleak\backends\winrt\client.py", line 769, in <lambda>
    characteristic, lambda: self._session.max_pdu_size - 3
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'max_pdu_size'

SimonJonsson1999, 2025-05-16:
Okay, can someone help me and have a look at the wireshark logs? I am not sure what to look for.
https://drive.google.com/file/d/1aYK-ZIkjuUueiZY7WDumcLQIGCQoynod/view?usp=sharing

dlech, 2025-05-16:
I don't know enough about the low level commands to say for sure what is wrong, but I can see that some of the low level (`LL_`) commands are happening in a different order compared to the working and not working case. So I suspect that this has something to do with Windows working in one case and disconnecting in the other.

For example, Wireshark flags this particular command as out of sequence. If Windows detects that too, that could be why it is disconnecting.

[image]

So unless you can modify the firmware on your device, I'm not sure what could be done in this case.

SimonJonsson1999, 2025-05-16:
Okay, I must thank you for your excellent support, I will close this issue.

closed 2025-05-16 by SimonJonsson1999