hbldh/bleak #1757: Bleak hangs on connect if another app has already established a BLE connection

plmn-tech, 2025-05-16:
**Environment**
- Windows 11 Enterprise Version 10.0.26100 Build 26100
- Bleak 0.22.3

**Problem**
Bleak hangs indefinitely when trying to connect to a BLE device that is already connected to another application.

**Steps to Reproduce**
1. Use another application (e.g., Bluetooth LE Explorer) to establish a BLE connection to the device that Bleak will attempt to connect to. (If possible, use a BLE packet sniffer to confirm that the connection has been successfully established)
2. Run your Bleak script to connect to the same device.
3. Regardless of whether you use the "`async with`" context manager or `client.connect()`, the connection attempt hangs indefinitely—timeouts do not work.

**Example Log Output**
```
2025-05-16 12:25:11,513 bleak.backends.winrt.client MainThread DEBUG: Connecting to BLE device @ xx:xx:xx:xx:xx
2025-05-16 12:25:11,987 bleak.backends.winrt.client MainThread DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2025-05-16 12:25:12,135 bleak.backends.winrt.client MainThread DEBUG: skipping service <uuid> due to access denied
2025-05-16 12:25:12,150 bleak.backends.winrt.client MainThread DEBUG: skipping service <uuid> due to access denied
2025-05-16 12:25:12,204 bleak.backends.winrt.client MainThread DEBUG: skipping service <uuid> due to access denied
2025-05-16 12:25:12,221 bleak.backends.winrt.client MainThread DEBUG: skipping service <uuid> due to access denied
```

The specific services that are denied access vary depending on which GATT services the other app is using.

**Minimal Reproducible Code**

```
device = await BleakScanner.find_device_by_address(address)
async with BleakClient(device, timeout=20.0) as client:
  print(connected")
```

or

```
device = await BleakScanner.find_device_by_address(address)
client = BleakClient(device)
await client.connect()
```

**Additional Notes**
If you reverse the scenario and let Bleak connect to the device first, other apps can still connect afterward and discover the GATT services without issues. The reference app is not limited to Bluetooth LE Explorer—I tested with three different apps and observed the same behavior in all of them.

plmn-tech, 2025-05-16:
Thanks @dlech ! The trace log is attached, it stuck at select -> poll

```
...
216516  --- modulename: __init__, funcname: filter
216517 __init__.py(827):         rv = True
216518 __init__.py(828):         for f in self.filters:
216519 __init__.py(836):         return rv
216520 __init__.py(1644):             self.callHandlers(record)
216521  --- modulename: __init__, funcname: callHandlers
216522 __init__.py(1700):         c = self
216523 __init__.py(1701):         found = 0
216524 __init__.py(1702):         while c:
216525 __init__.py(1703):             for hdlr in c.handlers:
216526 __init__.py(1707):             if not c.propagate:
216527 __init__.py(1710):                 c = c.parent
216528 __init__.py(1702):         while c:
216529 __init__.py(1703):             for hdlr in c.handlers:
216530 __init__.py(1704):                 found = found + 1
216531 __init__.py(1705):                 if record.levelno >= hdlr.level:
216532 __init__.py(1706):                     hdlr.handle(record)
216533  --- modulename: __init__, funcname: handle
216534 __init__.py(2216):     def handle(self, record):
216535 __init__.py(1703):             for hdlr in c.handlers:
216536 __init__.py(1704):                 found = found + 1
216537 __init__.py(1705):                 if record.levelno >= hdlr.level:
216538 __init__.py(1706):                     hdlr.handle(record)
216539  --- modulename: __init__, funcname: handle
216540 __init__.py(974):         rv = self.filter(record)
216541  --- modulename: __init__, funcname: filter
216542 __init__.py(827):         rv = True
216543 __init__.py(828):         for f in self.filters:
216544 __init__.py(836):         return rv
216545 __init__.py(975):         if rv:
216546 __init__.py(976):             self.acquire()
216547  --- modulename: __init__, funcname: acquire
216548 __init__.py(926):         if self.lock:
216549 __init__.py(927):             self.lock.acquire()
216550 __init__.py(977):             try:
216551 __init__.py(978):                 self.emit(record)
216552  --- modulename: __init__, funcname: emit
216553 __init__.py(1109):         try:
216554 __init__.py(1110):             msg = self.format(record)
216555  --- modulename: __init__, funcname: format
216556 __init__.py(949):         if self.formatter:
216557 __init__.py(950):             fmt = self.formatter
216558 __init__.py(953):         return fmt.format(record)
216559  --- modulename: __init__, funcname: format
216560 __init__.py(687):         record.message = record.getMessage()
216561  --- modulename: __init__, funcname: getMessage
216562 __init__.py(375):         msg = str(self.msg)
216563 __init__.py(376):         if self.args:
216564 __init__.py(377):             msg = msg % self.args
216565  --- modulename: uuid, funcname: __str__
216566 uuid.py(281):         hex = '%032x' % self.int
216567 uuid.py(283):             hex[:8], hex[8:12], hex[12:16], hex[16:20], hex[20:])
216568 uuid.py(282):         return '%s-%s-%s-%s-%s' % (
216569 __init__.py(378):         return msg
216570 __init__.py(688):         if self.usesTime():
216571  --- modulename: __init__, funcname: usesTime
216572 __init__.py(656):         return self._style.usesTime()
216573  --- modulename: __init__, funcname: usesTime
216574 __init__.py(433):         return self._fmt.find(self.asctime_search) >= 0
216575 __init__.py(689):             record.asctime = self.formatTime(record, self.datefmt)
216576  --- modulename: __init__, funcname: formatTime
216577 __init__.py(624):         ct = self.converter(record.created)
216578 __init__.py(625):         if datefmt:
216579 __init__.py(628):             s = time.strftime(self.default_time_format, ct)
216580 __init__.py(629):             if self.default_msec_format:
216581 __init__.py(630):                 s = self.default_msec_format % (s, record.msecs)
[... 90 lines cut ...]
216790  --- modulename: _weakrefset, funcname: __contains__
216807  --- modulename: proactor_events, funcname: _process_events
216811  --- modulename: base_events, funcname: time
216820  --- modulename: events, funcname: _run
216823  --- modulename: client, funcname: call_callbacks
216831  --- modulename: events, funcname: _run
216834  --- modulename: client, funcname: call_callbacks
216842  --- modulename: events, funcname: _run
216845  --- modulename: proactor_events, funcname: _loop_self_reading
216851  --- modulename: windows_events, funcname: recv
216853  --- modulename: windows_events, funcname: _register_with_iocp
216855  --- modulename: _weakrefset, funcname: __contains__
216865  --- modulename: windows_events, funcname: _register
216867  --- modulename: windows_events, funcname: _check_closed
216870  --- modulename: windows_events, funcname: __init__
216872  --- modulename: base_events, funcname: get_debug
216888  --- modulename: base_events, funcname: _run_once
216896  --- modulename: windows_events, funcname: select
216899  --- modulename: windows_events, funcname: _poll
216900 windows_events.py(813):         if timeout is None:
216901 windows_events.py(814):             ms = INFINITE
216902 windows_events.py(824):         while True:
216903 windows_events.py(825):             status = _overlapped.GetQueuedCompletionStatus(self._iocp, ms)  <<---- This is the last line of trace log

```

dlech, 2025-05-16:
Interesting, not what I was expecting. It looks like it is hanging in the core Python asyncio code (rather than on some WinRT function as I was suspecting).

But I think I know the problem now and may have already fixed it!

Can you try using Bleak from the develop branch and see if that solves the problem?

plmn-tech, 2025-05-16:
Hi @dlech, 

Thanks for the information! Unfortunately, the issue still persists with the develop branch. The trace log indicates that the logic is still getting stuck at select -> poll, as shown in the previous logs.
Since the commit you shared modified client.py, I captured a new trace log using the develop branch and filtered it by `client.py`. Hopefully, this provides some useful clues.

```
221100 client.py(1082):         return self._op.status != AsyncStatus.STARTED
221108 client.py(1131):         if not self.done():
221110 client.py(1082):         return self._op.status != AsyncStatus.STARTED
221118 client.py(1134):         return self.result()  # May raise too.
221120 client.py(1062):         if self._op.status == AsyncStatus.STARTED:
221128 client.py(1065):         if self._op.status == AsyncStatus.COMPLETED:
221136 client.py(1066):             if self._cancel_requested:
221137 client.py(1069):             return self._result
221138 client.py(745):                 if result.status == GattCommunicationStatus.ACCESS_DENIED:
221146 client.py(748):                     logger.debug(
221147 client.py(749):                         "skipping service %s due to access denied", service.uuid
221168 client.py(748):                     logger.debug(
221363 2025-05-16 15:55:13,260 bleak.backends.winrt.client MainThread DEBUG: skipping service <uuid> due to access denied
221391 client.py(751):                     continue
221392 client.py(742):             for service in services:
221393 client.py(743):                 result = await FutureLike(service.get_characteristics_async(*args))
221395 client.py(1042):         self._op = op
221396 client.py(1043):         self._callbacks = []
221397 client.py(1044):         self._loop = asyncio.get_running_loop()
221398 client.py(1045):         self._cancel_requested = False
221399 client.py(1046):         self._result = None
221400 client.py(1048):         def call_callbacks():
221401 client.py(1052):         def call_callbacks_threadsafe(op: IAsyncOperation, status: AsyncStatus):
221402 client.py(1059):         op.completed = call_callbacks_threadsafe
221411 client.py(1053):             if status == AsyncStatus.COMPLETED:
221412 client.py(1055):                 self._result = op.get_results()
221413 client.py(1057):             self._loop.call_soon_threadsafe(call_callbacks)
221447 client.py(1127):         if not self.done():
221449 client.py(1082):         return self._op.status != AsyncStatus.STARTED
221457 client.py(1131):         if not self.done():
221459 client.py(1082):         return self._op.status != AsyncStatus.STARTED
221467 client.py(1134):         return self.result()  # May raise too.
221469 client.py(1062):         if self._op.status == AsyncStatus.STARTED:
221477 client.py(1065):         if self._op.status == AsyncStatus.COMPLETED:
221485 client.py(1066):             if self._cancel_requested:
221486 client.py(1069):             return self._result
221487 client.py(745):                 if result.status == GattCommunicationStatus.ACCESS_DENIED:
221495 client.py(748):                     logger.debug(
221496 client.py(749):                         "skipping service %s due to access denied", service.uuid
221517 client.py(748):                     logger.debug(
221712 2025-05-16 15:55:13,277 bleak.backends.winrt.client MainThread DEBUG: skipping service <uuid> due to access denied
221740 client.py(751):                     continue
221741 client.py(742):             for service in services:
221742 client.py(783):             return new_services
221743 client.py(443):                 async with async_timeout(timeout):
221768 client.py(495):                 await event.wait()
221788 client.py(1049):             for c in self._callbacks:
221937 client.py(1049):             for c in self._callbacks:
221948 client.py(1049):             for c in self._callbacks:           <<---- note: the following lines to the end weren't filtered
221949 events.py(99):         self = None  # Needed to break cycles when an exception occurs.
221950 base_events.py(1920):         for i in range(ntodo):
221951 base_events.py(1921):             handle = self._ready.popleft()
221952 base_events.py(1922):             if handle._cancelled:
221953 base_events.py(1924):             if self._debug:
221954 base_events.py(1936):                 handle._run()
221955  --- modulename: events, funcname: _run
221956 events.py(83):         try:
221957 events.py(84):             self._context.run(self._callback, *self._args)
221958  --- modulename: proactor_events, funcname: _loop_self_reading
221959 proactor_events.py(790):         try:
221960 proactor_events.py(791):             if f is not None:
221961 proactor_events.py(792):                 f.result()  # may raise
221962 proactor_events.py(793):             if self._self_reading_future is not f:
221963 proactor_events.py(801):             f = self._proactor.recv(self._ssock, 4096)
[... 29 lines cut ...]
221993 proactor_events.py(814):             self._self_reading_future = f
221994 proactor_events.py(815):             f.add_done_callback(self._loop_self_reading)
221995 events.py(99):         self = None  # Needed to break cycles when an exception occurs.
221996 base_events.py(1920):         for i in range(ntodo):
221997 base_events.py(1937):         handle = None  # Needed to break cycles when an exception occurs.
221998 base_events.py(609):                 if self._stopping:
221999 base_events.py(607):             while True:
[... cut ...]
222009  --- modulename: windows_events, funcname: select
222010 windows_events.py(443):         if not self._results:
222011 windows_events.py(444):             self._poll(timeout)
222012  --- modulename: windows_events, funcname: _poll
222013 windows_events.py(813):         if timeout is None:
222014 windows_events.py(814):             ms = INFINITE
222015 windows_events.py(824):         while True:
222016 windows_events.py(825):             status = _overlapped.GetQueuedCompletionStatus(self._iocp, ms)
```

dlech, 2025-05-16:
Are you sure you are using the latest develop branch? The trace looks like it is from before the fix.

For example, `221768 client.py(495):                 await event.wait()` should be line 478 and should be more indented that the `async with async_timeout(timeout):` line above it.

plmn-tech, 2025-05-17:
Thanks a lot @dlech ! After I force-installed the develop branch, I confirmed the line number matched what you mentioned. and the connection was successfully established. 

Currently, the disconnect call always hits the 2mins timeout, which is expected because my bleak script didn't own that BLE connection. If possible, it would be helpful if bleak could detect when it's reusing a BLE connection opened by another app. In this scenario, it might not be necessary to wait 2mins to for a timeout. maybe an immediate exception like "Disconnect failed: connection owned by another app" would be more helpful

```
239468 client.py(711):             for service in services:
239469 client.py(768):             return new_services
239470 client.py(476):                     await event.wait()
239474 client.py(424):                 async with async_timeout(timeout):
239499 client.py(478):                 is_connect_complete = True
239500 client.py(480):                 self._services_changed_events.remove(services_changed_event)
239501 client.py(486):             self._session_active_events.remove(event)
239563 client.py(1061):             for c in self._callbacks:
239773 client.py(544):             if self._session is None
239774 client.py(545):             else self._session.session_status == GattSessionStatus.ACTIVE
239782 client.py(542):         return (
239788 client.py(490):         logger.debug("Disconnecting from BLE device...")
240004 client.py(492):         assert self.services
240005 client.py(495):         for handle, event_handler_token in list(self._notification_callbacks.items()):
240006 client.py(500):         self._notification_callbacks.clear()
240007 client.py(503):         if self.services:
240008 client.py(507):             await asyncio.sleep(0.1)
240218 client.py(509):             for service in self.services:
240223 client.py(510):                 service.obj.close()
240224 client.py(509):             for service in self.services:
240225 client.py(510):                 service.obj.close()
240226 client.py(509):             for service in self.services:
240227 client.py(510):                 service.obj.close()
240228 client.py(509):             for service in self.services:
240229 client.py(510):                 service.obj.close()
240230 client.py(509):             for service in self.services:
240231 client.py(510):                 service.obj.close()
240232 client.py(509):             for service in self.services:
240233 client.py(510):                 service.obj.close()
240234 client.py(509):             for service in self.services:
240235 client.py(510):                 service.obj.close()
240236 client.py(509):             for service in self.services:
240237 client.py(511):             self.services = None
240238 client.py(514):         if self._session:
240239 client.py(515):             self._session.maintain_connection = False
240240 client.py(522):         if self._requester:
240241 client.py(523):             event = asyncio.Event()
240245 client.py(524):             self._session_closed_events.append(event)
240246 client.py(525):             try:
240247 client.py(526):                 self._requester.close()
240248 client.py(529):                 async with async_timeout(120):
240311 client.py(530):                     await event.wait()
240474 client.py(529):                 async with async_timeout(120):
240482 client.py(532):                 self._session_closed_events.remove(event)

```

dlech, 2025-05-17:
I close this one since the original issue is solved and started a new issue for the disconnect problem.

closed 2025-05-17 by dlech