hbldh/bleak #1759: Windows does not close GATT session when another app is using the same peripheral

dlech, 2025-05-17:
> Currently, the disconnect call always hits the 2mins timeout, which is expected because my bleak script didn't own that BLE connection. If possible, it would be helpful if bleak could detect when it's reusing a BLE connection opened by another app. In this scenario, it might not be necessary to wait 2mins to for a timeout. maybe an immediate exception like "Disconnect failed: connection owned by another app" would be more helpful
> 
> ```
> 239468 client.py(711):             for service in services:
> 239469 client.py(768):             return new_services
> 239470 client.py(476):                     await event.wait()
> 239474 client.py(424):                 async with async_timeout(timeout):
> 239499 client.py(478):                 is_connect_complete = True
> 239500 client.py(480):                 self._services_changed_events.remove(services_changed_event)
> 239501 client.py(486):             self._session_active_events.remove(event)
> 239563 client.py(1061):             for c in self._callbacks:
> 239773 client.py(544):             if self._session is None
> 239774 client.py(545):             else self._session.session_status == GattSessionStatus.ACTIVE
> 239782 client.py(542):         return (
> 239788 client.py(490):         logger.debug("Disconnecting from BLE device...")
> 240004 client.py(492):         assert self.services
> 240005 client.py(495):         for handle, event_handler_token in list(self._notification_callbacks.items()):
> 240006 client.py(500):         self._notification_callbacks.clear()
> 240007 client.py(503):         if self.services:
> 240008 client.py(507):             await asyncio.sleep(0.1)
> 240218 client.py(509):             for service in self.services:
> 240223 client.py(510):                 service.obj.close()
> 240224 client.py(509):             for service in self.services:
> 240225 client.py(510):                 service.obj.close()
> 240226 client.py(509):             for service in self.services:
> 240227 client.py(510):                 service.obj.close()
> 240228 client.py(509):             for service in self.services:
> 240229 client.py(510):                 service.obj.close()
> 240230 client.py(509):             for service in self.services:
> 240231 client.py(510):                 service.obj.close()
> 240232 client.py(509):             for service in self.services:
> 240233 client.py(510):                 service.obj.close()
> 240234 client.py(509):             for service in self.services:
> 240235 client.py(510):                 service.obj.close()
> 240236 client.py(509):             for service in self.services:
> 240237 client.py(511):             self.services = None
> 240238 client.py(514):         if self._session:
> 240239 client.py(515):             self._session.maintain_connection = False
> 240240 client.py(522):         if self._requester:
> 240241 client.py(523):             event = asyncio.Event()
> 240245 client.py(524):             self._session_closed_events.append(event)
> 240246 client.py(525):             try:
> 240247 client.py(526):                 self._requester.close()
> 240248 client.py(529):                 async with async_timeout(120):
> 240311 client.py(530):                     await event.wait()
> 240474 client.py(529):                 async with async_timeout(120):
> 240482 client.py(532):                 self._session_closed_events.remove(event)
> 
> ``` 

 _Originally posted by @plmn-tech in [#1757](https://github.com/hbldh/bleak/issues/1757#issuecomment-2887937526)_

dlech, 2025-05-17:
I started a new issue since this is a separate problem.

> it might not be necessary to wait 2mins to for a timeout.

This exists because of issues that have been reported unrelated to having another app using the device, so I'm not keen on changing it.

> Disconnect failed: connection owned by another app

In Windows, we don't actually control literal connection and disconnection. We only get a GATT session. So it should just be waiting for our own GATT session to close.

It looks like in the `disconnect()` we only close the `BluetoothLEDevice` (`self._requester`) and the the GATT session (`self._session`) with the comment:

```python
            # calling self._session.close() here prevents any further GATT
            # session status events, so we defer that until after the session
            # is no longer active
```

IIRC, in the past, we tried not waiting and just closing the GATT session, but then reconnecting didn't work so well because Windows could wait a long time to disconnect the device before we could scan again.

So I'm not sure what we could do here without potentially messing up other users. I guess will have to try out a few different things and see what it does.

dlech, 2025-05-17:
I'm not sure how I could reproduce this issue though since I don't think I have any BLE devices that advertise while already connected.

plmn-tech, 2025-05-17:
Hi @dlech 

Thank you for following up on this disconnect timeout issue when the BLE connection opened by another app! 

By reading the title, I'd like to clarify that I didn't open any GATT services in my Bleak script — the code simply find_device_by_address, connect(device), sleep(1), and then disconnects.

I used Bluetooth LE Explorer to reproduce the issue when testing with devices like the iRobot and iPhone. or any device as long as the LE Explorer can connect to it. The disconnect call would hit the 2mins timeout if the BLE session is still active. The problem occurs if you start the script after entering the Device Services page in LE Explorer.

Please kindly note that sometimes iPhone only maintains the BLE session for about 40s, so you would see the disconnect happens soon

dlech, 2025-05-17:
I tried my headphones but I just got `bleak.exc.BleakError: Could not get GATT services: Unreachable` when trying to connect 2 apps at once. 

Out of curiosity, does this make a difference?

```diff
diff --git a/bleak/backends/winrt/client.py b/bleak/backends/winrt/client.py
index 571728f..5bd0a72 100644
--- a/bleak/backends/winrt/client.py
+++ b/bleak/backends/winrt/client.py
@@ -521,6 +521,7 @@ class BleakClientWinRT(BaseBleakClient):
             self._session_closed_events.append(event)
             try:
                 self._requester.close()
+                self._requester = None
                 # sometimes it can take over one minute before Windows decides
                 # to end the GATT session/disconnect the device
                 async with async_timeout(120):
```

plmn-tech, 2025-05-20:
Over this weekend, I found that there’s another scenario where connect() and disconnect() hang or timeout if the BLE Server is turned off and then back on (or power-cycled) during use. This behaves identically to the scenario where another app takes over the BLE connection first — as described in #1757 (connect) and #1759 (disconnect).

**Problem**
If the Gatt Server was turned OFF BT or powered cycle, the issues on connect() and disconnect() are exact same as the scenario that another app established the BLE first. I have confirmed from -trace, the code line where the logic hangs or timeout are same.

**Steps to reproduce**
1. export BLEAK_LOGGING=1 to view what uuid got access denied
2. Run the minimal test code below to establish a BLE connection
3. After seeing "Connected", turn off Bluetooth on the Gatt Server
4. Wait for the disconnect_callback() to be triggered
5. Turn BT back on and check the behavior on the 2nd connection attempt

**Test code**

```
import asyncio
from bleak import BleakClient, BleakScanner

address = "xx:xx:xx:xx:xx:xx"

SLEEP_TIME = 15.0

def disconnect_callback(client):
    print(f"Device {client.address} disconnected") # 2. turn back on the server's BT after seeing this

async def _connect():
    attempt = 0

    print(f"start")
    device = await BleakScanner.find_device_by_address(address)
    
    while attempt < 2:
        client = BleakClient(device, disconnect_callback)
        
        if client.is_connected:
            print(f"already connected")
            break

        try:
            await client.connect()
            print(f"Connected, sleep for {delay_time}s")  # 1 . turn off the server's BT after seeing this
            await asyncio.sleep(SLEPP_TIME)
        except asyncio.TimeoutError:
            print(
                f"Connection attempt timed out."
            )
        except Exception as e:
            print(f"Error during BLE operations: {e}")

        finally:
            if client.is_connected:
                try:
                    print_with_timestamp(f"try to disconnect")
                    await client.disconnect()
                except asyncio.TimeoutError:
                    print_with_timestamp(f"Disconnect times out")
                finally:
                    print_with_timestamp(f"Disconnected")
            else:
                print_with_timestamp(f"alreadty disconnected")
            
        attempt += 1
        print(f"end of attempt")

asyncio.run(_connect())
```

**Results**
connect()
1. if bleak is on the stable branch, the call will hang forever
2. if bleak is on the develop branch, the issue is resolved. (https://github.com/hbldh/bleak/commit/fc8fa2ad3ea2d1e88321e184aa79bc686d2d862b)
3. adding services=[] reduces "skip service uuid access denied" messages, but does not prevent the hang.
-  Without specifying services, Bleak logs show "Access Denied" for all service UUIDs after reconnecting.
-  With services=[<uuid>], only the specified UUID fails with "Access Denied"

4. If I re-use the same BleakClient instance across attempts, the 2nd connection works — even on the stable branch. (this is just a workaround I can use in simple test code but cannot be used in a real project, because the event loop might be changed at the 2nd connect, I got the exception of "Event loop Closed" in handle_value_changed.)

disconnect()
1. on the stable branch, this cannot be checked because connect() has been pending there.
2. on the develop branch, the call always hit the 2mins timeout (the topic we're talking at this page #1759)

I hope that this new scenario can bring more clues :-) . Please let me know if you have any questions, thank you!

dlech, 2025-05-20:
What happens if we skip the waiting and just close the gatt session in the disconnect method?

```diff
diff --git a/bleak/backends/winrt/client.py b/bleak/backends/winrt/client.py
index 1b48cb3..6b64fca 100644
--- a/bleak/backends/winrt/client.py
+++ b/bleak/backends/winrt/client.py
@@ -495,11 +495,14 @@ class BleakClientWinRT(BaseBleakClient):
                 self._requester.close()
                 # sometimes it can take over one minute before Windows decides
                 # to end the GATT session/disconnect the device
-                async with async_timeout(120):
-                    await event.wait()
+                # async with async_timeout(120):
+                #     await event.wait()
             finally:
                 self._session_closed_events.remove(event)
 
+        if self._session:
+            self._session.close()
+
     @property
     @override
     def is_connected(self) -> bool:
```

dlech, 2025-05-20:
> 4\. because the event loop might be changed at the 2nd connect

Unrelated to the issue, but as a side note, this sounds a bit odd. Generally, I would expect an app to have one run loop for the duration of the entire app.

plmn-tech, 2025-05-20:
> What happens if we skip the waiting and just close the gatt session in the disconnect method?
> 
> diff --git a/bleak/backends/winrt/client.py b/bleak/backends/winrt/client.py
> index 1b48cb3..6b64fca 100644
> --- a/bleak/backends/winrt/client.py
> +++ b/bleak/backends/winrt/client.py
> @@ -495,11 +495,14 @@ class BleakClientWinRT(BaseBleakClient):
>                  self._requester.close()
>                  # sometimes it can take over one minute before Windows decides
>                  # to end the GATT session/disconnect the device
> ---              async with async_timeout(120):
> ---                   await event.wait()
> +++                # async with async_timeout(120):
> +++               #     await event.wait()
>              finally:
>                  self._session_closed_events.remove(event)
>  
> +++        if self._session:
> +++           self._session.close()
> +++
>      def is_connected(self) -> bool:

Great! This patch fixes the disconnect timeout issue, I have verified both scenarios. Whether the BLE connection is held by another app or is disconnected by the server side, the `disconnect()` can be done without any delay, and I can confirm `self._session.close()` is properly called.

I'd like to bring up one more question related to the scenario where the server disconnects the BLE connection, Bleak consistently got service uuid "access denied" after reconnecting, and the behavior depends on what service were specified at the 1st connect.

> 3. adding services=[] reduces "skip service uuid access denied" messages
> -  Without specifying services, Bleak logs show "Access Denied" for all service UUIDs after reconnecting.
> - With services=[], only the specified UUID fails with "Access Denied"

The reason I'm highlighting this : if the denied service is the notify service we need to use, the 2nd connection becomes unusable. It looks like some Gatt related state isn't fully  cleaned up when the server disconnects the initial connection. 

I'd like to hear your thoughts on this - is there a recommended way for us to handle this case? Thank you!

plmn-tech, 2025-05-21:
> In particular, def handle_disconnect() is missing:    
>        for service in self.services:
>                 service.obj.close()

Thanks a lot for the response and clear explanation! Your solution for cleaning up service objects worked perfectly ! 

And I also tested calling `disconnect()` without checking `is_connected`, it worked as well even without the cleanup code placed in `handle_disconnect()` as you mentioned.  In the scenario we discussed, `disconnect()` might take up to 2mins, which's why I added the `is_connected` check. It's good to know that there's no harm in always calling `disconect()` to free resources even if the connection has already been torn down.

So far, we've discussed 3 issues and now have solutions for all of them:

1. `connect()` hangs
2. `disconnect()` hits a 2-minute timeout
3. service access denied on the 2nd connection

May I ask what your plan is for moving these fixes into the stable release? No rush at all — just checking in on the roadmap.

I appreciate your support and have a good day!

dlech, 2025-05-23:
I just merged the fix for 2. I think 3 I will defer until later since we have a workaround. I would like to really have a good look at this type of disconnection across all platforms and maybe reconsider how it works in general.

We're a bit overdue for a release, so hopefully we'll get one soon.

plmn-tech, 2025-05-23:
Thank you David for letting me know!