hbldh/bleak #1797: [ENHANCEMENT] `pythonista.cb` bleak backend

o-murphy, 2025-07-25:
I'm working on bleak compatible backend for pythonista iOS app, for now I done `BleakScanner` backend based on `pythonista.cb` and starts with `BleakClient`.

### What's working:
* `CentralManagerDegate`
* `BleakScanner.discover` by service uuid

### Work in progress
* `PeripheralDelegate`
* `BleakClient`

dlech, 2025-07-29:
Happy to look at a PR. I do have some concern that iOS support will be like Android support in that it is not quick/easy to test so it tends to get neglected.

o-murphy, 2025-07-29:
> Happy to look at a PR. I do have some concern that iOS support will be like Android support in that it is not quick/easy to test so it tends to get neglected.

This not close to Android support at the moment. Android support relies on the [kivy/p4a](https://github.com/kivy/python-for-android) project.
 
Whereas with iOS it is more like emulation or sandboxing due to Apple restrictions.
Thanks to [omz/Pythonista](https://github.com/omz/Pythonista), (although this project is proprietary), but it's the best existing python IDE for iOS.
It makes possible to use full Python interpreter and also pure python packages on iOS (which running in the Pythonista sandbox).

So Pythonista also supports part of the `ObjC API` and a wrapper around `CoreBluetooth` which is used in this backend, making it close to the existing CoreBluetooth backend for macOS.

Also I expect this to be supported in future [Pythonista Lab](https://omz-software.com/pythonista/lab/) app. It's now in TestFlight beta now

dlech, 2025-07-29:
If it is API-compatible with PyObjC, then that is probably not so bad as we wouldn't have to write a 2nd backend for CoreBluetoth.

o-murphy, 2025-07-29:
> If it is API-compatible with PyObjC, then that is probably not so bad as we wouldn't have to write a 2nd backend for CoreBluetoth.

Not really, the objc API is too limited in Pythonista. We have no direct support for CoreBluetooth, we have just a thin wrapper around it. 
That's why I developed a separate backend, it takes over some of the logic from the existing backend for macOS, but the implementation is very different.

Refer to [Documentation](https://omz-software.com/pythonista/docs/ios/index.html).
Pay special attention to [cb](https://omz-software.com/pythonista/docs/ios/cb.html) and [objc_util](https://omz-software.com/pythonista/docs/ios/objc_util.html)

o-murphy, 2025-07-29:
> If it is API-compatible with PyObjC, then that is probably not so bad as we wouldn't have to write a 2nd backend for CoreBluetoth.

If we had the ability to run the interpreter natively on iOS, the PyObjC option would most likely work, but currently there is no such option, but it won't work with Pythonista as it, only if the Pythonista developer is interested.
I will explore the possibilities of `ish` linux emulator for iOS, maybe there are differences

o-murphy, 2025-07-29:
Done in [#1799](https://github.com/hbldh/bleak/pull/1799)

timrid, 2025-07-31:
When talking about iOS and Python, BeeWare should not be forgotten. With [`briefcase`](https://github.com/beeware/briefcase), they offer a tool for developing iOS apps very easily. Thanks to the BeeWare team, iOS has now officially become part of CPython with [PEP 730](https://peps.python.org/pep-0730/).

Under the hood, `briefcase` uses [`rubicon-objc`](https://github.com/beeware/rubicon-objc) to do the Objective-C binding on iOS. This should make everything possible that is also possible in `PyObjC`. I wanted to look into this topic over the next few months to see if the macOS port can be easily adapted to iOS. But that will be a separate ticket. Pythonista seems to go into a complete different direction than BeeWare.

o-murphy, 2025-07-31:
> When talking about iOS and Python, BeeWare should not be forgotten. With briefcase, they offer a tool for developing iOS apps very easily. Thanks to the BeeWare team, iOS has now officially become [...]

I understand what you are talking about, but no one prevents you from supporting beeware and pythonista at the same time. The bottom line is that bleak is just an interface that envelops several native APIs.

I mean `bleak` itself is pure python and can be used anywhere in both native applications with both beeware like, p4a or sandbox as a pythonista if the environment offers a suitable backend.

Another point is that beeware needs xcode and building native application, which has its limitations for both developers and users. (For example, I don’t have a computer with macOS, and I have difficulties to debug native iOS apps.

While pythonista offers writing, editing and executing python code directly from iOS, without the need to have enough experience and technical base to create a native application with Python support. You just need to be familiar with just python. So I think `bleak` support for both options can be allowed.

By the way, I’m checking if bleak is running in pythonista, and not just on iOS, so it won’t be a problem to solve the backends for beeware/pythonista.

dlech, 2025-12-29:
This is listed in the docs under 3rd-party backends, so nothing left to do here.

closed 2025-12-29 by dlech