hbldh/bleak #1819: Access raw manufacturer data

himslm01, 2025-08-29:
I have a temperature / humidity device (TP357) which regularly announces its temperature, humidity and battery level data, which it puts into the manufacturer data part of its advertisement data .

I am unable to access the raw data because it is not being understood correctly by this section of code.

https://github.com/hbldh/bleak/blob/6b2c390fa55c924782bc5d2a3ace3e87322bf691/bleak/backends/corebluetooth/scanner.py#L125

The manufacturer data is not organised as the code thinks. This manufacturer data is just a byte array.

Please can I access the raw manufacturer data.

dlech, 2025-08-29:
According to the Bluetooth spec:

> The Manufacturer Specific data type is used for manufacturer specific data.
The first two data octets shall contain a company identifier from Assigned
Numbers. The interpretation of any other octets within the data shall be defined
by the manufacturer specified by the company identifier.

And explicitly defines "shall" to mean:

> used to express what is required by the specification and is to be implemented exactly as written without deviation

If you want to treat the company ID as data, you can convert it back to two bytes and prepend it to the rest of the data. 

Since this is not how devices are supposed to be using this field, so it isn't something that we intend to support directly.

closed 2025-08-30 by dlech

dlech, 2025-08-30:
> If that are "lots of hardware vendors" where this is a problem, isn't that a good argument to provide a simple way for users of the bleak library to get access to the raw data?

If I was a hardware vendor that paid money to the Bluetooth SIG to get a company ID and other vendors didn't respect that and made advertising data that had random data that happened to match my company ID and it interfered with the user experience of my product, I would be pretty angry about it. So, no, I don't intend to make it easy to do this as that would just encourage more people to not respect the intended use.

But the "lots" I was referring to is all sorts of bugs/violations of the spec, not just this specific one.