hbldh/bleak #1829: Windows: `BleakClientWinRT` `_get_services` frequently times out

gudnimg, 2025-09-19:
I've tried also tried extending the timeout to 30 seconds. No luck. I'm unsure how to debug this further. When I nRF Connect for Desktop or the nRF Connect app, discovering services is pretty quick. My peripheral is using Zephyr / NCS 3.1.0 firmware, not sure if that can affect things.

```
2025-09-19 12:12:40,060 bleak.backends.winrt.scanner DEBUG: 46 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
2025-09-19 12:12:40,174 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ [MY BDA]
2025-09-19 12:12:40,237 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2025-09-19 12:13:10,253 bleak.backends.winrt.client DEBUG: closing requester
2025-09-19 12:13:10,256 bleak.backends.winrt.client DEBUG: closing session
```

Some info about my setup:
* Windows 11 24H2 (26100.6584)
* Python 3.13.7
* winrt: 3.2.1
* pywin32: 311
* (More pip packages relevant?)

Code:

```python
# start the connection, here it hangs on discovering services
async with BleakClient( device, disconnected_handler ) as client:
    # Just some code not relevant to issue
```

I find the device using `await BleakScanner.find_device_by_name` and also sometimes `await BleakScanner.find_device_by_address`. I don't see a difference in behavior when it comes to discovering services.

----

I can debug further and share logs if possible. Any advice how to debug the issue further would be very appreciated! :)

gudnimg, 2025-09-19:
I've tracked the issue down to this wait in `BleakClientWinRT`. Services are actually discovered almost immediately.

```python
# a connection may not be made until we request info from the
# device, so we have to get services before the GATT session
# is set to active
await event.wait()
```

gudnimg, 2025-09-19:
Similar log except now it also includes the commands `HCI_CMD`. In this case it hangs for 12 seconds. [image]

gudnimg, 2025-09-19:
I wonder if this USB event is related to the issue `URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL`. I'll try and find time to test this on another computer over the weekend.

[image]

gudnimg, 2025-09-19:
`URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL` went away went I disabled the option to allow Windows to power down the Bluetooth dongle. Issue still persists though, the time needed to wait for the connection is quite random. Sometimes 12 seconds, sometimes over 30 seconds.

dlech, 2025-09-19:
> I haven't used this tool before, I'll see if I can create a better log for you :)

You can do a File > Save in Wireshark and put the file in a .zip and attach it to a comment here on GitHub.